iNavFlight/inav #601: Support separate builds for different NAZE board revisions?

martinbudden, 2016-09-13:
Perhaps we should produce separate builds for the different NAZE boards, ie rev4, rev5 rev6 etc.

The reason for this is we could save ROM space by only including the required GYRO and ACC. I would suggest except ROM space is really tight on the NAZE.

Any view anyone?

DzikuVx, 2016-09-13:
I have mixed feeling. That way we can multiply number of targets we support ... and what about different clones, Flip32, SkyLines etc...

martinbudden, 2016-09-13:
I've just tried a build and it saves about 2.5K of ROM. Which isn't enough to put LED_STRIP back in.

So it's probably not worth the effort of supporting all those different builds. As you say, there are loads of variants.

digitalentity, 2016-09-13:
It would be probably be better to have builds w/o servos (for quads only) and with servos (tri's and airplanes). Separation of airplane code might be the other option.

xdigyx, 2016-09-14:
best would be to compare options vs memory saved

oleost, 2016-09-14:
What we really need is built in compiler in the configurator, thats user friendly.

So when flashing we get to choose exactly what we want, and which version to compile (latest master, latest development, latest release) etc etc.

I know someone have to make it, and thats the issue.

But atleast maybe we should open a new issue regarding this and reach out to the community ? This is something that would benefit all forks of cleanflight.

stronnag, 2016-09-14:
Rather a build server. Installing a build environment for multiple operating systems is not practical. 

This is how rapid flash works for simonk ESCs.

xdigyx, 2016-09-14:
End user (web?) configuration based compiler would be best option, but for sure would need lots of work and time. Who can make it? And what until that time: is there enough resources or as a first step  some faster solution should be implemented?

martinbudden, 2016-09-14:
We should probably think of starting to use Docker at some point anyway.

@oleost , you're right this is something we should do. But as you say, someone needs to do it. I'm keen to keep supporting legacy hardware, but this would still be quite a lot of work.

So, in the short term at least, the builds with and without servers seems to be the easiest fix.

digitalentity, 2016-11-15:
postponing this, again...

digitalentity, 2016-12-26:
#960

closed 2016-12-26 by digitalentity