iNavFlight/inav #603: feature proposal: Advanced headfree mode

andre92, 2016-09-13:
The way headfree mode is currently working is that it uses the the takeoff heading (set when arming) as the head direction for the controls. This works ok if you always keep the craft in front of you and arm it while facing away from you.

Ardupilot and MuliWii have an advanced version of this mode, where it uses a head direction relative to the home direction, once you pass a certain distance from home (ADV_HEADFREE_RANGE in multiWii)

The way this works in MultiWii is to set headfreemodehold to GPS_directionToHome + 180 every loop, if the distance to home is greater than adv_headfree_range, and if headfree mode is active. setting adv_headfree_range to 0 will disable the feature. This could simply be added to the headfree code, and we only need one extra cli variable.

I would like to hear some thoughts on this.

rb1205, 2016-09-15:
This would be nice. While flying in FPV i have no use for headfree, however a Home-bound version would be useful for those cases where you wish to go back home without heading straight to it.

DzikuVx, 2016-09-15:
And isn't it just better to engage Return-To-Home ?

krzysztofmatula, 2016-09-15:
I think this feature could be useful not only for emergency.
It is similar to DJI "Home Lock", where pitch moves the aircraft further/closer and roll circles around the home point without changing distance. Extending this with the possibility to set the center point (in flight) would allow to fly smooth circles around some object being filmed. It's hardly achievable with today's modes with and without head lock.

martinbudden, 2016-09-16:
I don't think this would be difficult to implement.

DzikuVx, 2016-09-16:
@krzysztofmatula @martinbudden I was thinking about it and looks like all has to be done, is to always set heading to home. With CRUISE mode that should more less give circling pattern.

closed 2018-05-27 by stale[bot]