iNavFlight/inav #606: PID tuning for GPS hold

courierdude, 2016-09-15:
Hi all. My first post so I hope I'm doing this right. 
QAV540G
Naze32 full
Ublox m8n
When in GPS hold, the quad wanders up to about 10 meters. Was showing signs of long slow toilet bowling so I've re calibrated compass (on the naze board). Can someone point me to a link where I can research how to tune the GPS PIDS to try and lock this baby in? I've searched and searched but can't seem to find the info I need. 
Thankyou

DzikuVx, 2016-09-15:
Default GPS Hold PIDs are good and solid, usually there is no need to retune it.
If I have to guess I would tell, that the reason is onboard magnetometer too close to power cables, ESCs or other electrical devices. Built in magnetometers are extremely sensitive and in 95% of cases they are the reason UAV can not keep position.
Blackbox log would be very helpful to confirm that.
rearrange cables, but better use external magnetometer far away from all metalic/electric devices

courierdude, 2016-09-17:
Thank you so much for replying. I am running a Naze32 f/c and just purchased a http://www.hobbyking.com/hobbyking/store/__99119__Mini_Ublox_NEO_M8N_GPS_for_Naze32_Flip32.html. 
But it doesn't appear to have the compass wired which is why I'm still using the compass on the Naze32. I've moved the f/c as far as possible from the power wires, but obviously not far enough. Any suggestions if it's possible to shield the compass from interference?

bk79, 2016-09-17:
> Ublox_NEO_M8N_GPS_for_Naze32_Flip32.html.

Put some aluminium foil with heat shrink around esc/wire and solder a wire
on it connected to the negative of the battery plug

courierdude, 2016-09-17:
Thanks, I'll give it a try. Does any one know if you can identify interference of the compass through iNav GUI? I see there's a moving graph but I don't know how to decipher the read out. It is looking pretty steady even after I throttle up with no props of course. If it was getting interference, I would have expected it to be all over the show.

bk79, 2016-09-17:
Easy way is to verify on the main window if north corresponds to 0 heading
with everything off then move to the graph tab and read the values for x y
z, now tie it so it cannot move from where it is so to keep always 0
heading and north, install props and run motors then go and read the values
again they probably are different. If heading and north is not shown...it
means you have a trouble.....an easier way to test it make a run in ph if
it makes toilet bowls it means you have a problem with mag

courierdude, 2016-09-17:
Excellent. Thank you. I'll check it out.

stronnag, 2016-09-17:
Black box log. fly straight lines at reasonable throttle. Post the logs. Someone here (me, digitalentity, others) will analyse it for you and check the correlation between mag direction and heading from GPS.

courierdude, 2016-09-17:
Great. Thank you. Raining here at the moment, not due to clear for several days. I'll do it as soon as possible.
Actually now I think of it, I've flown a few times and I think the Black box has been recording. Can you point to a link so I can learn how to get the info?

stronnag, 2016-09-17:
Not sure there are any links. I use a script (from mwptools/samples) that correlates heading (attitude[2]/100) against calculated course from consecutive  GPS locations where the speed is greater than some value (say 4m/s) and throttle above some value (say 1500). The arbitrary values to help ensure we are in steady flight where the GPS will give reasonable directional information (which is not the case in hover).

MarcoPTG, 2016-09-22:
Hello Guys. My copter after iNav 1.2 become a little unstable on PosHold and i am trying to improve the GPS PID Values but its a little bit hard but i will keep trying. The RTH is working Fine but need to improve the NavR PID Values to be more stable. But the iNav need the Autotune for PID values. I didn't test the Mag Mode and if i can help on this count me in. I am a newbie on this. 
PS: Do you guys know if the Fail Safe RTH is working? My doesn't work. Only Fail Safe Land.
Thanks.
MSilva.

closed 2016-11-21 by digitalentity

blakowl, 2017-07-17:
Can you explains the Pos, Posr, and Navr abbreviations in the GPS tuning pids please

oleost, 2017-07-17:
Use rcgroups thread for support. Also there are some brief explanations on in the cli.md document.

This issue is closed, please continue on rcgroups thread.

georgekucher, 2017-11-12:
Hi!
Should I configure the GPS PIDs if I have old BT_GPS [module | http://www.mondoplast.ro/eng/Bluetooth-GPS-receiver-West-Studio-BT-77-pg_ft-2655] which runs at 1Hz and 16 channels on NMEA protocol? The problem I'm facing with is, not even bowling but fast running from side to side. When I'm enabling PH mode, after couple of seconds the quadcopter rapidly runs aside for near 5 meters, then to the opposite direction but more far, and so on in 15-30 meters radius of the point where PH was turned on.
The MAG is external and mounted 10-15 cm below the FC and other wiring (excepting the main BAT, its in 5 cm of it). I'll upload a video with OSD to be sure.
Sadly I have Flip32 (F1 NAZE) board and can't have a blackbox logging.

Or maybe this old GPS is a crap and I should throw it to trash?)))

georgekucher, 2017-11-12:
[Here](https://youtu.be/R_-d7vSzAW4) it is.
Sorry for quality ;)

digitalentity, 2017-11-13:
1Hz GPS is not enough for INAV to function properly.

georgekucher, 2017-11-13:
Should I try to switch GPS velocity? May it help?

digitalentity, 2017-11-13:
You can try configuring your GPS for 5Hz update rate - it may help. Changing settings likely won't help.

georgekucher, 2017-11-13:
Specs says that 1Hz is max ((

georgekucher, 2017-11-13:
Please look is everything else settings is ok

# INAV/NAZE 1.8.0 Nov 10 2017 / 20:15:28 (912d1315)
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature GPS
feature CURRENT_METER
feature LED_STRIP
feature PWM_OUTPUT_ENABLE
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACTION_SUCCESS
beeper ACTION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB
beeper LAUNCH_MODE
beeper CAM_CONNECTION_OPEN
beeper CAM_CONNECTION_CLOSED
map AETR5678
name -
serial 0 1 115200 38400 0 115200
serial 1 2 115200 38400 0 115200
led 0 4,6::C:0
led 1 5,6::C:0
led 2 6,6::C:0
led 3 7,6::C:0
led 4 8,6::C:0
led 5 9,6::C:0
led 6 10,6::C:0
led 7 11,6::C:0
led 8 0,7::CW:2
[... 334 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 40
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

digitalentity, 2017-11-13:
Config looks ok to me. With proper GPS you should be good to go.

georgekucher, 2017-11-13:
What do you think about [this one](https://m.banggood.com/7M8M-Ublox-M8N-GPS-Module-for-APM-Pixhawk-CC3D-Naze32-F3-Flight-Control-p-1078657.html)

digitalentity, 2017-11-13:
Nah, that's a quite crappy module. 
I'd go for a trusted BN-880 https://www.banggood.com/UBLOX-NEO-M8N-BN-880-Flight-Control-GPS-Module-Dual-Module-Compass-p-971082.html
Alternatively there's also this module which I prefer to any other so far https://www.banggood.com/Ublox-NEO-M8N-Flight-Controller-GPS-with-Protective-Shell-for-PIX-PX4-Pixhawk-p-1005394.html

georgekucher, 2017-11-13:
Thank you!
But the sad thing is that I've already ordered it on Saturday(((

blakowl, 2017-11-13:
> Studio-BT-77-pg_ft-2655] which runs at 1Hz and 16 channels on NMEA

I was having major trouble, never could get the GPS hold to work.  It
turned out it was my GPS receiver. I ordered this one and it worked
perfectly.

Usmile Beitian NEO-M8N GPS Module with Built-in HMC5883L Compass Protective
Case for Ublox PX4 PIX APM2.5/2.6 Pixhawk
https://www.amazon.com/dp/B01EH3819W?ref=yo_pop_ma_swf

georgekucher, 2017-11-13:
Which GPS did you use when you had a trouble?
Was that the one I ordered?
https://m.banggood.com/7M8M-Ublox-M8N-GPS-Module-for-APM-Pixhawk-CC3D-Naze32-F3-Flight-Control-p-1078657.html

blakowl, 2017-11-13:
> Pixhawk-CC3D-Naze32-F3-Flight-Control-p-1078657.html

No it was a neo6m

georgekucher, 2017-11-13:
Oh! Ok, so fingers crossed))

georgekucher, 2018-05-28:
Hey guys!
Finally I got a newer board (Matek F722-STD) and I still can't get my quad a GPS hold.
A GPS module seems to work fine. I think my mag QMC5883 is not working properly.
@digitalentity could you please have a look on blackbox logs?
I've even combined them with video to see more details.
p.s. I'm using cam gimbal, so rolls and pitches can be seen only on OSD bars on video.

Here is my dump:
[dump.txt](https://github.com/iNavFlight/inav/files/2046253/dump.txt)

Logs with PH attempts:
https://drive.google.com/file/d/1rp2x7gbO5cZelk0bwZj7FsBdij4XzTdE/view?usp=sharing
https://youtu.be/KpQk_ELnrug

Logs with yaws:
https://drive.google.com/file/d/15t4xv0xar2PE8ynqmY3m4DC4KfuCyEVk/view?usp=sharing
(video tomorrow morning)

georgekucher, 2018-05-29:
Sorry for confusing. All work now. Just reconfigured from the very begin (from flashing the board) step by step. I think the problem was that I was calibrating accelerometer with mag enabled (just a guess). This time I did the accelerometer calibration first, then I've enabled the MAG and calibrated it as well. No drifting any more. Pos Hold works. Woohhooo!