iNavFlight/inav #607: strange behaviour during front/back flips in 1.2

gurkenfolie, 2016-09-15:
I recently flashed official 1.2 release, now i have the problem that in horizon mode the quad levels really fast after 360 flips/rolls (but the quad should still turn as i want correlating with stick input). i set transtion to 10, so i can fly almost acro. worked with 1.1 like a charm.
the next problem is when i do pitch rolls/flips, the quad behaves strange, it doesnt turn correctly as it did with 1.1., it does as strange movement which is hard to describe, it rolls a little bit during pitch flip.
Please take a look at my bb log, i can provide further logs if necessary. i dont know why it behaves so strange...
in acro mode everything works as it should. tried this on naze32 rev6 and spracing f3, same results. rollback to inav 1.1 solved the issue.
tried it with 1.2 RCimnotsure and 1.2 stable.

the last log is pitch flip only.

[SEQLOG00.TXT](https://github.com/cleanflight/cleanflight/files/474835/SEQLOG00.TXT)
[dump.txt](https://github.com/cleanflight/cleanflight/files/474839/dump.txt)

digitalentity, 2016-09-16:
@gurkenfolie 
1.2 adopted OpenPilot's Rattitude logic for Horizon mode. Looks like it's not behaving correctly when transition is low (like 10% in your case) and self-leveling is not completely disabled.

I'll make a change, hopefully today, and would greatly appreciate if you test it (at the moment I'm traveling w/o my quadcopter).

digitalentity, 2016-09-16:
@gurkenfolie please check this hex - it now makes sure that beyond set transition point you have no self-leveling whatsoever.
[inav_1.2.1_NAZE.hex.gz](https://github.com/iNavFlight/inav/files/476131/inav_1.2.1_NAZE.hex.gz)

oleost, 2016-09-16:
Same issue?

From Rcgroups:
AmiSMB: "Issue explained: With iNav 1.2 in Horizon mode I can roll 100% left or right and the quad only rolls to around 45 degrees left or right without doing a full rotation which it did with Cleanflight and Betaflight. No GPS modes enabled. In the blackbox log you can see on flight 2 at between 10s and 20s I try full left and full right roll."

http://www.rcgroups.com/forums/showpost.php?p=35736770&postcount=8180

AmiSMB, 2016-09-16:
I have found the same issue where full pitch or roll input does not allow the craft to flip. I have posted in RCGroups with the setup and logs which can be found by clicking [here](http://www.rcgroups.com/forums/showthread.php?t=2495732&page=546#post35736770)

stronnag, 2016-09-16:
Very strange. Horizon flips and rolls work for me, with default PIDs (iirc). On all frames, bit ugly on the big tri. Have done on all versions, somewhat ndistinguishable from clean/betaflight. Rate is 550 deg/sec on the big frames and probably higher on the 250. I'll confirm these later.

digitalentity, 2016-09-16:
It was probably ok  with LEVEL D (Horizon threshold) at high value (i.e. default 75) with relatively high rates and at full stick deflection. However at low setting of LEVEL D and RC stick not reaching full range of 1000-2000 values one may be experiencing a bug with self-leveling not being completely disabled. 

Possible effects:
 - not being able to do flips/roll at all - self-leveling will prevent that from happening
 - drift on roll/pitch axis when performing a flip/roll.

stronnag, 2016-09-16:
I flew the new horizon-mode-changes branch today, with my "brute force and ignorance" it seems (LOS) "mainly harmess" in that horizon mode rolls and flips were as awesome as before.

This build exhibits yet more mag problems (complete failure to hold heading in PH mode). Unfortunately for me, I'd neglected to clear out the SPIFLASH prior to flying, rendering an hour of structured testing across multiple builds nugatory.  So I'll have the tedium of doing it again before generating a new issue against 1.2.1.

digitalentity, 2016-09-16:
Good! Thanks for testing. We really need to look into mag issues.

stronnag, 2016-09-16:
yes, sorry for no logs but some interesting videos in #611. You make, I break it.

gurkenfolie, 2016-09-16:
thx DE, can you give me a hex for spracing f3? otherwise i would have to change the FC...

stronnag, 2016-09-17:
SPRF3 hex http://seyrsnys.myzen.co.uk/inav_737c7d0/inav_1.2.1_SPRACINGF3.hex

gurkenfolie, 2016-09-17:
thanks, i will try it ASAP

gurkenfolie, 2016-09-19:
build the newest inav version for spracingf3 , everything worked well except PH, it was toiletbowling for one circle when changing the position. after one circle it settled. (mag is calibrated...)

digitalentity, 2016-09-19:
@gurkenfolie HORIZON mode works, we can close this issue.

PR #608 (2016-09-18): HORIZON mode changes
- [x] Make sure self-leveling is completely disabled above transition point (fixes #607)
closed 2016-09-19 by digitalentity