iNavFlight/inav #609: Performance drop in 1.2 (new IO?)

DzikuVx, 2016-09-16:
While preparing to test #421 I've noticed that 1.2 has much lower performance than 1.1

SPRacingF3
In 1.1 async gyro/pid/acc was able to run at 2.6kHz gyro and 666Hz PID
in 1.2 top performance is 2kHz gyro and 500Hz PID

Somehow I suspect this is new IO / i2c related. @martinbudden @digitalentity  any ideas?

Yes, I2C in both cases is "overclocked" and setup exactly the same
`set i2c_overclock = ON`

martinbudden, 2016-09-16:
I don't think it's related to the conversion of the drivers to the new IO, since the code there is common with betaflight and we would certainly have heard if there had been lower performance in betaflight.

So I suspect it is changes in the PID code. @digitalentity have there been any changes there that might have affected performance? (I haven't really looked at the PID code for a while.)

DzikuVx, 2016-09-16:
I think I have it:

`i2cSetOverclock(masterConfig.i2c_overclock);`
then, 
```
void i2cSetOverclock(uint8_t overClock)
{
    i2cOverClock = overClock ? true : false;
}
```
but 
`i2cOverClock` is never read.
@martinbudden please take a look, this is slightly too deep in hardware for me expertise

martinbudden, 2016-09-16:
@DzikuVx , that could be it. I'll have a look later on.

digitalentity, 2016-09-16:
@DzikuVx, @martinbudden 
Confirmed, overclock setting was not used for initialising the I2C bus. Will submit a fix in a minute.

digitalentity, 2016-09-16:
#610 should fix the problem

DzikuVx, 2016-09-16:
I'm closing it, PR in place

commit 9030f1b2 (2016-09-16): Fix for I2C overclock not used. Closes #609 (#610)
closed 2016-09-16 by DzikuVx