iNavFlight/inav #611: PH yaw highly unstable in 1.2.1

stronnag, 2016-09-16:


digitalentity, 2016-09-16:
Very interesting indeed. Looks like mag is either gone completely wrong or got disabled (if read will fail IMU will ignore the mag data).

stronnag, 2016-09-16:
The mag is not disabled (the 'leds' in the bottom right are sensor status). The reported heading is perfectly correct --- I know this field very well. The mag is working correctly, as reported over LTM. 

The the videos are two of s series of experiments over an hour. 
1.2.0 PH yaw always stable, with or without 0f63bc6,
1.2.1 Always rotated counter-clockwise, but the reported heading direction was geospatially correct.

digitalentity, 2016-09-16:
Hm... If possible, can you verify MAG mode?

stronnag, 2016-09-16:
yes, and I'll explicitly enable it with PH.

digitalentity, 2016-09-16:
Interestingly enough, current 1.2.1 is not that different from 1.2.0 - I'm puzzled.

stronnag, 2016-09-16:
Join the puzzled club. I did not expect this ....

stronnag, 2016-09-17:
Here's some more information. There is a blackbox log at http://seyrsnys.myzen.co.uk/inav_121_ph/BBLOG_155023.TXT with a variety of 1.2.0 and 1.2.1 logs in it.

Index: 1 fw: INAV 1.2.1 (737c7d0) SPRACINGF3 
Index: 2 fw: INAV 1.2.1 (737c7d0) SPRACINGF3
Index: 3 fw: INAV 1.2.1 (737c7d0) SPRACINGF3
Index: 4 fw: INAV 1.2.0 (a5f7067) SPRACINGF3
Index: 5 fw: INAV 1.2.0 (a5f7067) SPRACINGF3
Index: 6 fw: INAV 1.2.1 (737c7d0) SPRACINGF3
Index: 7 fw: INAV 1.2.0 (a5f7067) SPRACINGF3

For the purposes of this report, Indices 1 and 5 are not interesting.

Index 2, 1.2.1 : PH 'toilet bowl' and yaw variation.
Index 3: 1.2.1  : PH 'toilet bowl' and yaw variation (patched to remove  0f63bc6)
Index 4: 1.2.0 :  Stable PH, insignificant yaw variation. MAG enforced on PH (as for all subsequent tests)
Index 6: 1.2.1 : PH 'toilet bowl', no yaw variation (MAG enforced).

From index6 and the 10m range circle, the PH diameter is c. 2m, compared  with c. 0 in 1.2.0.

Index 2 (1.2.1)
[image]

Index 3: (1.2.1)
[image]

Index 4 (1.2.0 yeh!!!, ph circle so small you can hardly see it)
[image]

Index 6 (1.2.1 PH forced MAG)
[image]

Hardware is a 250mm dead cat aka 'Cecil the Lion', (custom mix), Neo6M, BMP180, HMC5883. CLI Dump at //seyrsnys.myzen.co.uk/inav_121_ph/dc_sprf3_fp.txt

Some further tests were run on my 420mm quad (the #431 test bed), whilst they are not as extreme, behaviour under 1.2.1 is a noticeable regression v. 1.2.0, and a report on that platform will follow soon.

digitalentity, 2016-09-17:
@stronnag thanks for testing. I'll try to have a look at the logs soon.
Did you try to `git bisect` the problem on development branch?

digitalentity, 2016-09-17:
@stronnag just a thought:
The most significant thing that changed in the code is filtering - @martinbudden did some work porting notch filter code (see daf9ba7ddef6da91d52123b53801f01b60344500 and #592). I wonder if there's a bug in the code that affects flight performance.

stronnag, 2016-09-17:
I will look into testing a reverted daf9ba7 / #592 build tomorrow.

digitalentity, 2016-09-17:
Specifically, I have a hunch about this change:
https://github.com/iNavFlight/inav/commit/daf9ba7ddef6da91d52123b53801f01b60344500#diff-340f0f98312ec72859f73ed36f86423dR52

Navigation code assumes that `dT` may vary and keeping that in mind the code is using RC-filter which allows that.

It's very high probability that at very first PH cycle filtering code is called with `dT` smaller than actual position estimate update rate:
https://github.com/iNavFlight/inav/blob/master/src/main/flight/navigation_rewrite_multicopter.c#L404
and
https://github.com/iNavFlight/inav/blob/master/src/main/flight/navigation_rewrite_multicopter.c#L445-L446

Effect of that is much lower cutoff frequency and much bigger delay in PH response.

digitalentity, 2016-09-17:
Before @martinbudden used cached `dT` in calculating the RC-filter first cycle with small `dT` value didn't cause us any trouble, but now it's affecting overall performance.

Personally, I think we shouldn't use cached `dT` in `pt1FilterApply4` which by it's name is expected to use the `dT` value that is passed to the function.

digitalentity, 2016-09-17:
@stronnag don't revert the https://github.com/iNavFlight/inav/commit/daf9ba7ddef6da91d52123b53801f01b60344500, instead apply this patch - 1f8e360

I have a strong belief that it will solve the problem.

stronnag, 2016-09-17:
OK, I have un-reverted daf9ba7 and will apply 1f8e360 and try again tomorrow --- it's wine o'clock here now.

digitalentity, 2016-09-17:
@stronnag I forgot that you are at GMT time zone :smile: 
Good night!

stronnag, 2016-09-17:
It's not late, just a pleasant Romanian Pinot Noir! Anyway, would you prefer I test an unadulterated pt1-filter-change branch or patch 1f8e360  into the horizon-mode-changes branch. I can do either.

digitalentity, 2016-09-17:
Sorry, with all those time zones, jet lags and traveling I got completely
lost in time :)

Horizon code shouldn't affect PH performance, you can test any variant
which is more convenient! I hope the filtering patch will solve the issue
completely.

stronnag, 2016-09-17:
no problem. I appreciate the time zone issue, I'm so happy I'm not doing the mega business travel thing any more.  I have the pt1-filter-change branch ready to go.

martinbudden, 2016-09-17:
I made the change to the filter because of the desire to share the code with betaflight more than anything else. I sort of assumed `dT` remains constant, but I guess that is an invalid assumption. If the tests show it causes the problem I'll make a PR to change it back.

stronnag, 2016-09-17:
@martinbudden, no problem. The real problem is that testing struggles to keep up with the rate of change (which is only going to get worst as the N hemisphere days shorten). Still, at least the unique (for CF derivatives) saving of acc / mag configuration makes in-field flashing and testing practical.

martinbudden, 2016-09-17:
We have been through a phase of code refactoring, which included adding the new IO and F4 support. It's not quite finished - there's still parameter groups to go, and there's also some minor refactoring of the receiver code I want to do.

But after that I hope to spend my time on adding new stuff - so hopefully things will stop breaking and the testing requirement will be less.

stronnag, 2016-09-17:
well if you run out of things to do, there's always the wonderful βF 'diff' and 'diff all' CLI options, which I think would have a lot of iNav uses a lot of trouble.

stronnag, 2016-09-18:
I flew many flights (PH, RTH, WP) today on the pt1-filter-change branch and it fixes the problem. Perfect nav performance again. Please merge.

digitalentity, 2016-09-18:
Fix merged

PR #614 (2016-09-18): Remove caching of dT value from pt1FilterApply4
Fixes #611
closed 2016-09-18 by digitalentity