iNavFlight/inav #620: Quad turns to north after arming and takeoff with mag enabled

oleost, 2016-09-19:
From @carlonb on rcgroups. Link to post http://www.rcgroups.com/forums/showpost.php?p=35752674&postcount=8199

Both CLI dump and blackbox record is included in post over at rcgroups.

Quote carlonb:

Hi all,
After many happy flight with iNav 1.1 now I've updated to iNav 1.2 but with a strange result.
At very first power on (batt connection), after arm, I can hear the motors with higher rate than 1.1, and as soon I lift it up, the copter turn facing north. After this, the yaw control became as usual.

As I land, disarm and arm again (without powering off), the rotation to north will not happen anymore also the motors turn as usual at minimum rate after arm. 

Have you noted this ? Any suggestion ?
Here my log and cli dump.
First log: Power on - rotation to north
Second log: Normal behavior after disarm and re-arm
Third log: Same normal behavior in Pos-Hold

Thanks
Bye, Carlo

carlonb, 2016-09-19:
Hi, I want to specify that I'm using the SPRacingF3 original (Mag on board) as a FC.
Used that with iNAV 1.1 for  a couple of moths with no issues, just yesterday I've updated with iNAV 1.2 (obviously same h/w) and I saw this strange thing.
Carlo

xdigyx, 2016-09-20:
Can you recalibrate MAG and check it few times?
every time before powering, face it to a different direction

carlonb, 2016-09-20:
OK, understood, I will try your suggestions, but cause my job I will test it on next week end.
Thanks
Bye

oleost, 2016-09-20:
@carlonb As far as I understood there is no issue with both RTH and Poshold?

Then I find it unluckely there is anything wrong with his calibration of mag, and sounds like some issue that it does not reset mag hold position.

Is there anyone else that succesfully arm with poshold and mag hold? I never done it before.

digitalentity, 2016-09-20:
Correct me if I'm wrong - you have MAG mode always enabled, from the moment
of connecting battery?

carlonb, 2016-09-20:
@digitalentity 
Correct, I have always enabled the mag, so yes, mag is already enabled when I power on the copter
@oleost 
Yes Poshold work very fine, I've not tried RTH as I saw this problem. Anyway, with iNav1.1 Poshoil and RTH ran very well.

digitalentity, 2016-09-20:
Ok, then I know what's the issue is. At power up heading is initialized at
zero (north) and it takes a while to converge. We need to delay locking the
heading in MAG mode until the copter is armed.

carlonb, 2016-09-20:
Ok, if you can compile a beta for me (I'm not able to do it by myself) i can test it.

digitalentity, 2016-09-20:
The fix is not existent yet, stay tuned :)

carlonb, 2016-09-20:
Of course, I do not want it now, I can test it in next week end.
Thanks.

MarcoPTG, 2016-09-22:
Hello Guys. My copter after iNav 1.2 become a little unstable on PosHold and i am trying to improve the GPS PID Values but its a little bit hard but i will keep trying. The RTH is working Fine but need to improve the NavR PID Values to be more stable. But the iNav need the Autotune for PID values. I didn't test the Mag Mode and if i can help on this count me in. I am a newbie on this. 
PS: Do you guys know if the Fail Safe RTH is working? My doesn't work. Only Fail Safe Land.
Thanks.
MSilva.

carlonb, 2016-09-23:
News after more tests iNav 1.2:
What I said in the first comment is not quite accurate, so read the following please.

Very first copter power on with MAG always active:
1) Arm, Take-Off, the copter always turn facing true North (if arm nose direction is not NORTH), after this, normal behavior.

2) After land, disarm, rearm, take off,  the nose direction behave as normal.
3) After land, disarm, MANUALLY I change the nose direction (some degree left or right), re-arm, take off, the copter always turn and reach the last landing/disarm direction, after this normal behavior.

Hope this can help.
Bye Carlo.

DzikuVx, 2016-11-04:
@carlonb could you retestit with INAV 1.3 ? There were maghold mode changes that probably fixed it

carlonb, 2016-11-04:
OK, I will test ASAP. Thanks

carlonb, 2016-11-06:
Tested 1.3.0 just now, this issue seems solved! :)).
Test done:
Power on with MAG always active:
1) Arm, Take-Off, the copter no more turn to North (Solved).
2) After land, disarm, rearm, take off, the nose direction behave as normal.
3) After land, disarm, MANUALLY I change the nose direction (some degree left or right), re-arm, take off, the copter no more turn to the last landing/disarm direction (Solved).

Tested also activating MAG after take off, all seems run well, poshold also run well.

For my point of view all run OK.
Thanks guys.

digitalentity, 2016-11-06:
Thanks for testing, closing.

closed 2016-11-06 by digitalentity