iNavFlight/inav #621: Sonar originated altitude is (almost) always lower by few cm

jflyper, 2016-09-20:
As I was testing the sonar support for OMNIBUS F3, I came to notice that the distance reported by sonar is always shorter than the actual distance by 1~3cm.

I took a quick look at the code, and found at least a couple of places where loss of significant digit is occurring (accounts for 2cm).

I actually don't know if sonar is requested to generate a distance in this precision.

Any thoughts?

Oh, and it came to me that we should have a variable to adjust the altitude based on the relative position of where the sonar is mounted on a vehicle.

bk79, 2016-09-20:
Consider that sonar is reliable below 3m and baro is 0.25 m precise... 2 cm
is quite precise for me

jflyper, 2016-09-20:
Okay. Then I would look into the occasional jump; 350cm at 10cm and a like.

bk79, 2016-09-20:
Yes those glitchesvare something to look into personally i would use baro
to have a reliabilty threshold

jflyper, 2016-09-20:
Shall we open a new issue?

jflyper, 2016-09-23:
@bk79 @martinbudden 

What do you think of this? betaflight/betaflight#1212

bk79, 2016-09-23:
The idea of having a disabled interrupt micros() could probably solve the
issue did you have a test?

martinbudden, 2016-09-23:
Interesting. Perhaps sonar should used `SysTick->VAL` directly and handle rollover itself?

jflyper, 2016-09-23:
@bk79 
No, I just picked this up this morning.

@martinbudden 
May be a good idea.

martinbudden, 2016-09-23:
@jflyper sonar range is max 4m. So the max time for a sonar return pulse is 2* 29.4 * 400 = 23520 microseconds (sound travels at about 29.4 cm/us). So we should only ever get 1 rollover.

jflyper, 2016-09-23:
@martinbudden 
What if a falling edge is missed for some 'really bad' reason?

bk79, 2016-09-23:
3.5m max...over that tested on arduino you get rubbish and if you don't
have a smooth surface or you have a soft sound absorbing one...garbage as
well...so probably it's better to limit it at 2 m

martinbudden, 2016-09-23:
Yes, limiting to less than 4m makes sense. Because the valid range for a pulse time is small, it should be fairly straightforward to detect missed echoes and ignore them.

jflyper, 2016-09-24:
Okay, so it looks like the micros() called from ISR may be 1msec (little less) older than the actual timestamp. It explains occasional 30+cm spikes in the sonar measurement; 1msec older timestamp at rising edge, true timestamp at falling edge = total 1msec longer pulse.

Still, I recall I have seen 300cm+ measurement at 10cm actual. I wonder if I mis-read the 30+ as 300...

digitalentity, 2016-09-24:
You didn't misread anything, sonar is very susceptible to vibration and noise. I've been playing with sonars for quite a while and I was unable to find one that will behave on a miniquad - all of them seem to read garbage after certain throttle value.

stronnag, 2016-09-24:
I was offered (for a significant sum) a LightWare SF11/C today (laser rangefinder to 120m). Interesting device, but currently at a cost of more than I've ever spent building a 'copter. Unlikely to be tempted.

jflyper, 2016-09-25:
Ah, I can understand that; vibe are velocities.

Time for optical?

bk79, 2016-09-25:
260$ for halfing the baro precision...mmm...probably not worth

jflyper, 2016-09-26:
LIDARs are overkill.

There's an IR-LED based 0-15m solution at around $115 
[LeddarOne](http://www.robotshop.com/en/leddartech-leddarone-optical-rangefinder-33v-uart.html
), but it still is overkill.

I think I'm going to try combine several cheap IR-LED based devices;

- 100cm-500cm GP2Y0A710K0F
http://www.sharp.co.jp/products/device/doc/opto/gp2y0a710k_e.pdf
(There's ambiguity below 100cm)

- ~200cm
[VL53L0X Time-of-Flight Distance Sensor Carrier with Voltage Regulator, 200cm Max](https://www.pololu.com/product/2490)

martinbudden, 2016-09-26:
There's also the Teraranger http://www.robotshop.com/en/teraranger-one-tof-rangefinder-type-a-spider.html

jflyper, 2016-09-28:
@martinbudden Look nice. Got into my list of optical range finders.

digitalentity, 2016-09-29:
Optical is nice, however I still don't have the time to improve the sonar logic in INAV. Probably after #543

jflyper, 2016-10-11:
#677 should improve things a little bit, if not at all.

I still want to evaluate vib effects digitalentity mentioned, though.

Ah, and some optical stuff, may be.

Closing for now.

closed 2016-10-11 by jflyper

digitalentity, 2016-10-12:
@jflyper yes, #677 should fix issues with timing, but poor support of sonar altitude is deeper than that. At the momemt my time has beed diverted to better supporting fixed wing aircraft.