iNavFlight/inav #622: Stage 1 failsafe throttle setting needs change to work with nav_use_midrc_for_althold = 1

rb1205, 2016-09-20:
When the FC triggers failsafe stage 1 a smart thing to do is to set the aux channels to engage POSHOLD and ALTHOLD modes to stop the multicopter wherever it is, so that it can wait peacefully the timeouts for the failure to clear or to initiate RTH/landing/whatever.

However, this can't be achieved if nav_use_midrc_for_althold is set to 1 because there's no way to force the throttle to midrc during failsafe: it can be either set to "auto" (low and not working with RTH) or to the last valid value received (VERY unsafe). We need either to make the Auto policy smarter or to extend the "set" policy to throttle as well, so that one can issue a "rxfail 3 s 1500" to force throttle to mid.

oleost, 2016-11-04:
"to hold the last valid value received (VERY unsafe in this case as could cause a drop or a flyaway). We need either to make the auto policy smarter or to extend the "set" policy to throttle as well, so that one can issue a "rxfail 3 s 1500" to failsafe throttle to center."

I can't see how this is safer than hold.

With these proposed settings the aircraft will spin up motors if failsafe engages while it's armed.

With the hold function it will stay at current throttle, so if your in air it will stay and on bench / ground not spin up.

oleost, 2016-11-04:
Also, stage 1 is very short and should in neither cases cause a fly away.

rb1205, 2016-11-04:
> With the hold function it will stay at current throttle, so if your in air it will stay and on bench

Not necesarly true. If you're flying "aggressively" or even freestyling your throttle could be any position from 0% to 100%. In those cases you could gain (or lose) up to 10 meters in a second, expecially if you set the FC to engage leveled flight.

> With these proposed settings the aircraft will spin up motors if failsafe engages while it's armed.

Just barely: there's a little spike of motor power when the mode kicks in, but the FC realizes soon that the height doesn't change and goes back to little more than idle throttle.

> Also, stage 1 is very short and should in neither cases cause a fly away.

Yeah that's true, i should probably rephrase it in "uncontrolled rise in height".

I dunno, I agree that the "hold" approach is good enough for the typical racing quadcopter that cleanflight aims to, but for inav i feel like it's lackluster. We have GPS, we tipically have larger (and more expensive) quads. Also, I think that failsafe stage 1 could be used as a user-tweakable configuration to wait for GPS signal before triggering emergency landing (see #733)

closed 2017-01-30 by digitalentity