iNavFlight/inav #625: Naze32 wont arm when Nav features would be programmed

WernerSch, 2016-09-22:
Hello

i have a flip32+ and inav 1.2.  I have programmed the Nav Modes RTH and Position Hold to a switch on my Sender. When i do this i cant arm my Copter. No Mode of both are activ. I get outside to get a GPS fix. GPS works fine and the inav App shows the Position on the map. But i dont can arm my Copter. 
I delete both Nav modes. After that i can armed my Copter.
I dont understand this. GPS works buts i cant use it. Somebody can help?

kind regards
Werner

DzikuVx, 2016-09-22:
Please provide screenshot of Cobfigurator Modes tab and or CLI dump

WernerSch, 2016-09-22:
Hello

here the dump and the Screenshot.

2016-09-22 @ 14:30:13 -- Running - OS: Windows, Chrome: 53.0.2785.116, Configurator: 1.0.2
2016-09-22 @ 14:30:13 -- You are using an old version of INAV - Configurator. There may be a more recent version with improvements and fixes.
2016-09-22 @ 14:30:18 -- Serial port successfully opened with ID: 1
2016-09-22 @ 14:30:18 -- MultiWii API version received - 1.21.0
2016-09-22 @ 14:30:19 -- Flight controller info, identifier: INAV, version: 1.2.0
2016-09-22 @ 14:30:19 -- Running firmware released on: Sep 7 2016 11:26:35
2016-09-22 @ 14:30:19 -- Board: AFNA, version: 1
2016-09-22 @ 14:30:19 -- Unique device ID received - 0x672ff575650877067032633
2016-09-22 @ 14:31:08 -- CLI mode detected
Setup
Ports
Configuration
Failsafe
PID Tuning
Receiver
Modes
Adjustments
Servos
GPS
Motors
Race Transponder
LED Strip
Sensors
Tethered Logging
Blackbox
CLI
Note: Leaving CLI tab or pressing Disconnect will automatically send "exit" to the board. With the latest firmware this will make the controller restart and unsaved changes will be lost.

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/NAZE 1.2.0 Sep  7 2016 / 11:26:35 (6cc415a)
# pflags
# Persistent config flags: 0x00000001
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
[... 303 lines cut ...]

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

# 
![22-09-2016 14-30-43]

(https://cloud.githubusercontent.com/assets/22211324/18748208/7e26c114-80d1-11e6-8159-558574870d41.jpg)

[image]

the NAV modes are disabled at moment. Now i can armed the copter. Before i use AUX3 for enable NAV POSHOLD & NAV RTH

kind regards
Werner

oleost, 2016-09-22:
How many gps sats do you have locked in? You need 6 or more,  and 3d fix

WernerSch, 2016-09-22:
ok oleost. This i must check tomorrow. Now its dark in germany :-)

I test inav today without GPS function and have now the next Problem. I calibrate the sensors and leraned on the ESC. But the motors turn irregular. It was a horror. I have Afro ESC 30A and BlHeli
Should i make a new issue?
Can i post here a video, so its the best to see the problem

kind regards
Werner

oleost, 2016-09-22:
New issue with as much detail as possible if your are sure it's a bug. 

If you believe it can be user error go to the forum and ask, also here provide as much info as possible. The wiki gives you a template for asking for help.

WernerSch, 2016-09-23:
Thats it :-) Thanks for help. After 7 Sats the Copter can be armed. 
But when i test the Copter on my terrace i have only 4 Sats. Then i can not armed it and make some tests. Must i always disable the GPS feature in the inav App before?

kind regards
Werner

rb1205, 2016-09-23:
This is a bug tracking service, not a support forum. Head over to the [forum ](http://www.rcgroups.com/forums/showthread.php?t=2495732) and ask for support there.

oleost, 2016-09-23:
No you don't, there is a setting disabling this extra safety. 

Read Wiki and please close this issue.

digitalentity, 2016-09-23:
@WernerSch please consider reading the wiki.

closed 2016-09-23 by digitalentity