iNavFlight/inav #627: Naze32 motors working irregular

WernerSch, 2016-09-23:
Hello

I have a Flip32+ and inv 1.2 on my copter. All is in 4S. I have Afro ESC 30A with BlHeli and Motory from Turnigy. But the Motors are working irregular. When i control one motor via the inav App its the same but i think not so extremly
Look at the video, so you can see it best
https://youtu.be/KydcZbRq6dg
I have no Idea why.

kind regards
Werner

bk79, 2016-09-23:
Video is private

oleost, 2016-09-23:
As I mentioned in the other issue was if you certain it's a bug,  then go
here. Else go to the forum.

I also directed you to the wiki regarding tips on what info you should
thinking about providing when reporting an issue or asking for help on the
forum.

There are key information missing in this bug report.

Please go to the wiki and read!

(especially needed are did it work on iNav 1.1? Does it work in
cleanflight? How are you doing this? And much more)

digitalentity, 2016-09-23:
I would guess a hardware issue, unrelated to firmware but it's hard to tell. "Irregular" doesn't really define anything.

@WernerSch please collect intormation @oleost suggested and post it to forum or here.

closed 2016-09-23 by digitalentity

WernerSch, 2016-09-23:
Sorry for the missunderstanding
I look now in the Forum RC Groups but there are 550 pages :-( 
Shoul i post there my problem at the end of page 550?

kind regards
Werner

commit b40b35e1 (2015-03-29): Updating GPS tested hardware list.  Closes #627