iNavFlight/inav #629: Fixed Wing: Extremely erratic altitude reporting with GPS

DzikuVx, 2016-09-24:
Since my airplane is operational, let's do some FW testing :)

* INAV 1.2
* Flip32 with Baro and Mag
* NEO-6M -> fix with 5-7 sats

Symptoms: with GPS fix, altitude reported over MSP to OSD (not sure about internal data since not using PosHold/AltHold on FW yet) was extremely erratic: from -50m to 100m in few seconds. 

I suspect it's a result of extremely poor fix and the fact that FW uses GPS for altitude and vertical velocity. Probably this is fine with good GPS module and fix.

@digitalentity I suspect setting `w_z_gps_p` and `w_z_gps_v` to `0` would force INAV to use BARO for altitude, but this is hard to do without extremely good knowledge of software. Maybe some kind of a CLI swith to simplify the thing?

digitalentity, 2016-09-24:
There is a bug in GPS/BARO combined handling of altitude, https://github.com/iNavFlight/inav/pull/543 is a work in progress to improve things. Basically current code is unable to handle altitude correctly if BARO and GPS are misaligned and they likely are.

As a workaround one can zero out `w_z_gps_p` and `w_z_gps_v` exactly as you suggested, or zero out `w_z_baro_p` and `w_z_baro_v` to leave only GPS as a reference.

oleost, 2016-09-24:
Or as the wiki suggest ;) 

Disable barometer (set baro_hardware = 1) on Naze32 targets due to critical bug that can make airplane dive to ground.

DzikuVx, 2016-09-24:
@oleost I prefer to use baro over GPS altitude. Somehow I do not trust GPS altitude too much

oleost, 2016-09-24:
Sure it's better with baro, please report back if zeroing w_z_gps_p and w_z_gps_v does the trick. Then I can update the wiki to reflect that instead.

sppnk, 2016-09-24:
I have to say I am flying several planes without baro (CC3D and nazes) with great success. Better to get the baro code working, but not needed.

digitalentity, 2016-11-17:
Hope to test over the weekend

digitalentity, 2016-11-20:
This should be fixed by #543

digitalentity, 2016-11-22:
Still unable to test. If somebody can flight-test this (GPS + BARO enabled) and provide a log it would be awesome.

oleost, 2016-11-22:
I was supposed to test it the day I was out, but forgot to enable barometer :/

Will test as soon as weather permits.

digitalentity, 2016-11-22:
@oleost no worries! Also no need to engage any navigation modes, just recording a simple log will do just fine.

digitalentity, 2016-11-27:
@oleost any change you would be able to test this in the nearest future?

oleost, 2016-11-27:
I hope so.

Just waiting for weather. My wing and battery is charged and ready

oleost, 2016-11-27:
@digitalentity Looking at the weather forecast I either get a chance tomorrow if not there will be heavy wind for one week.

digitalentity, 2016-11-28:
@oleost, @DzikuVx  I'm closing this for now - looks fine on the bench. Don't have the time to flight-test it here, but I doubt it will misbehave in flight.

If you encounter this bug again - feel free to reopen.

closed 2016-11-28 by digitalentity

oleost, 2016-11-28:
@digitalentity https://drive.google.com/open?id=0B-yHzNtOXxbEa181WGstVTRhTGs

Log, CLI dump and video.

I think I noticed some strange altitude loss while circle above home with RTH and first try.

Have not have timed to look at the log myself yet.

oleost, 2016-11-28:
@digitalentity 

There if deffinitly something wierd here. First time doing RTH on this wing. Please have a look.

[image]

Height at start:
[image]

Height at end:
[image]

Keep in mind that my baro is not shieldet in any way as I was not planning to use it. Except that its below one the flightcontroller.

digitalentity, 2016-11-29:
Ok, looks like it's not a BARO issue, but a GPS issue. During that weird altituce change GPS altitude drops from 40m to -22m (negative altitude):

[image]

Baro on the contrary indicate correct behavior (rise in altitude) when airplane climbs in attempt to compensate. Ok, the initial issue is indeed solved - BARO/GPS are synchronised, GPS is preferred, however the altitude estimation itself may be wrong. I wonder if we should favor barometer instead of GPS?

oleost, 2016-11-29:
Hmm. How is it on multirotors? (But I would assume that if you have BARO enabled it should be correct and be the preffered way of measuring height, on both multirotors and fixedwing)

But it is wierd, granted I dont have flown this fly very much, but I have done POS/ALT hold without any issues previously. In gps only mode. Wonder why this happend now or if its only concidence.

martinbudden, 2016-11-29:
> I wonder if we should favor barometer instead of GPS?

From what I've read GPS altitude is less reliable than baro altitude, so favouring barometer makes sense.

DzikuVx, 2016-11-29:
I'm for favoring baro over GPS altitude

digitalentity, 2016-11-29:
See #842