iNavFlight/inav #631: MAG (internal or external) Not detected

gantonjo, 2016-09-25:
Good evening.

Thanks for your efforts in making the iNavFlight.

I have a [10DOF SpRacing F3 Clone from Banggood  with internal compass](http://www.banggood.com/Upgrade-NAZE32-F3-Flight-Controller-Acro-6-DOF-Deluxe-10-DOF-for-Multirotor-Racing-p-1010232.html) and in addition I have an [Ublox Neo-7M GPS/Compass from Hobbyking](https://www.hobbyking.com/hobbyking/store/uh_viewItem.asp?idProduct=55558). The latter I used on my HKPilot 2.7 FC but now I have moved that to my newer platform.

Problem is that neither the internal compass on the F3 FC, nor the external compass in the Neo-7M is detected (connected to I2C bus). I have tried iNavFlight 1.2.0 and 1.2.1 to same result. 

To verify that this was not caused by HW error, I flashed the F3 FC with TriFlight 0.5 and there all sensors were detected and working. Going back and forth between iNavFlight 1.2.x and TriFlight 0.5 show same result. iNavFlight does not detect the compasses.

Thanks in advance for your help in getting the compass work, preferably the external one as it is on a pole.

Best regards Gantonjo

jflyper, 2016-09-28:
@gantonjo 
Do you know what the type of mag devices are used for internal and external compass?

gantonjo, 2016-09-28:
@jflyper 
According to the spec, the SPRF3 clone has a HMC5883 compass internally, and the external one in my GPS should be a HMC5883l digital compass. Whithout knowing anything about these, I would guess both compasses are equal.

jflyper, 2016-09-28:
@gantonjo 
It they are both HMC5883s, then you should remove one of them (presumably the internal one).

This is a separate problem from the detection problem you have with 1.2 and later.

They have the same address, so they respond simultaneously to read and writes over the I2C bus. Because I2C bus is open collector/open drain bus, any device that pull the bus low wins. It is usually the case if two different device with the same address are present on a bus, they compete for the bus and the result is a chaos.

For your case, you have two instances of the same device, and they may be detected because they respond to identification reads in the same way. However, once they are access for compass reads, you can expect less precision, because you will loose 1s from a device if another device write 0s on a particular bit and vice versa. Differences in bits are more likely in less significant bits, thus loss of precision there.

gantonjo, 2016-09-28:
Thnanks, @jflyper .
Hmm... I am not sure about your answer, because even without the external compass connected, the internal one is not detected by iNav.

jflyper, 2016-09-28:
@gantonjo 
Yeah, twin compass problem is orthogonal to the detection problem.

digitalentity, 2016-09-29:
INAV 1.2+ has better sensor error detection. I believe many would prefer compass to be not detected on the ground than to misbehave in flight.

gantonjo, 2016-10-01:
I accept the reasons for improving sensor detection, if there where faults in it. But, "if it ain't broke, don't fix it".

Now I have done a test with only the onboard mag connected  (that is, external mag and GPS is disconnected). Here is the status, bootlog and dump:

`# status
System Uptime: 27 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 0.02
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611
Cycle Time: 2013, I2C Errors: 1, config size: 2004

# resource
IO:
----------------------
A00: PPM IN
A01: FREE 
A02: FREE 
A03: FREE 
A04: ADC BATTERY
A05: ADC CURRENT
A06: FREE 
A07: FREE 
A08: LED STRIP OUT
A09: SERIAL1 UART TX
A10: SERIAL1 UART RX
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: FREE 
B00: SOFTSERIAL2 UART RX
B01: SOFTSERIAL2 UART TX
B02: FREE 
B03: LED1 OUT
B04: SOFTSERIAL1 UART RX
B05: SOFTSERIAL1 UART TX
B06: I2C1 SCL
B07: I2C1 SDA
B08: FREE 
B09: FREE 
B10: FREE 
B11: FREE 
B12: FLASH CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C13: MPU EXTI
C14: FREE 
C15: BEEPER OUT
F00: FREE 
F01: FREE 
F03: FREE 
F04: FREE 

# bootlog
Time Evt            Description  Parameters
   0:  0          CONFIG_LOADED 
   0:  1       SYSTEM_INIT_DONE 
 500: 19   TIMER_CHANNEL_MAPPED  (0, 0, 0, 0)
 500: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 500: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 500: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 500: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (7, 6, 1, 3)
 501: 19   TIMER_CHANNEL_MAPPED  (8, 6, 2, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (9, 6, 2, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (10, 6, 2, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (11, 6, 2, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (12, 6, 2, 3)
 501:  2          PWM_INIT_DONE 
[... 455 lines cut ...]
set gyro_soft_lpf_hz = 60
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500
`

digitalentity, 2016-10-18:
Is the problem still valid with 1.3-RC1?

gantonjo, 2016-10-18:
Thanks for your answer @digitalentity. I will check this later this week, perhaps already tonight if time permits.

gantonjo, 2016-10-18:
Hi, @digitalentity et al. Thank you for your efforts with iNavFlight.
Tonight I tried to flash the 1.3-RC1 to my SPRF3 clone. First try did not reveal any improvements. Then I tested another board I had laying around. I had TriFlight on that, but due to baro not showing up on that board I was expecting it to be faulty. Well, that board did show mag (both in TriFlight and in iNavFlight 1.3-RC1). One thing I decided to test on this board was to remove the foam I had on the baro. Voila! There the baro came to life!

Then I removed the foam from the SPRF3 mentioned in my first posts . Now I am able to see both the GPS, baro and mag. How to see if it is the Internal mag or the mag in the UBLOX GPS I am not sure. Here you see the status and bootlog:
`# bootlog
Time Evt            Description  Parameters
   0:  0          CONFIG_LOADED 
   0:  1       SYSTEM_INIT_DONE 
 501: 19   TIMER_CHANNEL_MAPPED  (0, 0, 0, 0)
 501: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (7, 6, 1, 3)
 501: 19   TIMER_CHANNEL_MAPPED  (8, 6, 2, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (9, 6, 2, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (10, 6, 2, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (11, 6, 2, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (12, 6, 2, 3)
 501:  2          PWM_INIT_DONE 
 547:  9         GYRO_DETECTION  (2, 0, 0, 0)
 663: 10          ACC_DETECTION  (3, 0, 0, 0)
 708: 11         BARO_DETECTION  (3, 0, 0, 0)
 708: 12          MAG_DETECTION  (2, 0, 0, 0)
2367: 13  RANGEFINDER_DETECTION  (0, 0, 0, 0)
2367:  4       SENSOR_INIT_DONE 
2918:  5          GPS_INIT_DONE 
2918:  6     LEDSTRIP_INIT_DONE 
2968:  7    TELEMETRY_INIT_DONE 
3036:  8           SYSTEM_READY 

# status
System Uptime: 138 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 0.04
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 2012, I2C Errors: 2, config size: 1988
`
At the moment the weather is way to bad for testing the TriCopter (RCExplorer V4-ish) outside. Anyway, seeing the sensors again make me optimistic for a better experience. I will maybe even flash the iNavFlight that has been modified with TriFlight improvements at a later stage. Look forward to having the TriFlight performance with added iNavFlight Nav performance;-) I maybe even will build a quad to test pure iNavFlight performance, just to learn what it can and cannot do;-)

Again, thanks for your great efforts.

digitalentity, 2016-10-19:
Looking forward to your findings.

digitalentity, 2016-10-19:
Can we close the issue?

gantonjo, 2016-10-19:
@digitalentity, you may close this issue. I did see some glitches in the detection of the mag after I answered you yesterday, but I believe that is due to something in my hardware. Thanks again.

closed 2016-10-19 by digitalentity