iNavFlight/inav #632: hmc5883l mag not detected until reboot in 1.2 and 1.2.1

thegreen4me, 2016-09-26:
First let me say I was upgrading from inav 1.1 and there were no issues with detecting this magnetometer in that version.  In both 1.2 and 1.2.1 I have a problem where the magnetometer is not detected when powering up the drone.  If I reboot the cc3d by CLI without disconnecting the battery, it will recognize the magnetometer when the cc3d comes back on and it will continue to recognize it perfectly fine until powering off the drone.

DzikuVx, 2016-09-26:
Looks like more of #631 #594 #552 

On my side this weekend I've tested build-in HMC5883l on Flip32 and external on SPracingF3 and had no problems with them

jflyper, 2016-09-28:
Interesting. This one looks like that there isn't enough delay after power on reset.

@thegreen4me 
I assume you are using an external mag, possibly integrated with a GPS module. Which one is that?

thegreen4me, 2016-09-28:
Yes, it is the compass integrated with this GPS unit here:
http://www.banggood.com/UBLOXNEO-M8NBN-880-Flight-Control-GPSModuleDual-Module-p-974361.html

Also an interesting side note, rebooting only makes it work in iNav 1.2. In iNav 1.2.1 rebooting does not solve the problem however sometimes once in a while it will work in 1.2.1 on a cold boot.  In 1.2 it never works on a cold boot but always works as soon as you reboot by CLI. Again, it always works 100% of the time in iNav 1.1

jflyper, 2016-09-28:
Similar symptom has been reported by @anbello in #578 (7th comment or so).

digitalentity, 2016-09-29:
I suspect it's related to poor quality HMC5883 chips. We might want to consider increasing the boot delay to give the chip time to wake up.

martinbudden, 2016-09-29:
@digitalentity , @DzikuVx , I think we need to look at the compass initialisation a bit more - there are still some issues it appears.

DzikuVx, 2016-09-29:
Personally I see nothing wrong with increasing boot delay. It's not like its taking a minute or so :)

wilco1967, 2016-09-29:
Perhaps not the same, but at least similar.

On my RCexplorer F3FC board, the external compass always initializes just fine, but the **baro** (5611) sometimes fails to initalize. usually, a power cycle resolves it. 
Both compass and baro are on I2C. The gyro / ACC are on SPI.
This is version INAV/RCEXPLORERF3 1.2.0 Sep 15 2016 / 16:25:20 (a5f7067)

on previous version, (15 sept / 2a7b4d2) it always worked flawless.

Once it boots with a working baro, it will keep working just fine throughout the flight, without any further I2C errors. I usually do get 1 I2C error only on boot (same with earlier versions).

jflyper, 2016-09-30:
@wilco1967 
Isn't 2a7b4d2 newer than a5f7067 ?
>  @kangsterizer kangsterizer committed 15 days ago    1 parent a5f7067 commit 2a7b4d2a598c58e40d2092e8ff343e36333689e8

wilco1967, 2016-10-01:
Oops, Sorry,

actual version which has the trouble is INAV/RCEXPLORERF3 1.2.1 Sep 22 2016 / 19:58:05 (36954a8) (just read back from the copter)
I must have mixed up my backup files before... Sorry....

wilco1967, 2016-10-01:
downloaded the current development version,
# INAV/RCEXPLORERF3 1.2.1 Oct  1 2016 / 18:32:11 (2b4018e)

have booted up at least 10 times now, and never any failure in detecting sensors... So it seems to work now... Don't know what changed, but it's ok now

haven't flown it yet, but the sensors seem to work just fine.

stronnag, 2016-10-01:
That is not development branch (1.2.2):
````
# INAV/SPRACINGF3EVO 1.2.2 Oct  1 2016 / 20:19:26 (9ff90ab)
````
 (2b4018e) is lastest master (stable).

wilco1967, 2016-10-01:
huh ?
I just got it of github this evening...

Or does git pull get the master instead, even though when called from the development directory ?

stronnag, 2016-10-01:
````
git checkout development
git pull
make TARGET=foo
# etc
````

wilco1967, 2016-10-01:
aha.....  lets try that... thanks !

but anyway, that (2b4018e) does seem to work fine for sensors....

stronnag, 2016-10-01:
and 
````
git checkout master
````
outs you back on stable, just in case you get bored of living dangerously. (Actually, dev is boringly safe at the moment, but I'm sure someone  will take care of that in due course).

wilco1967, 2016-10-01:
INAV/RCEXPLORERF3 1.2.2 Oct  1 2016 / 20:41:30 (9ff90ab)
This seems development, correct ?

I still have trouble understanding how github works.... 
just followed the information on: https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows.md 

just having a look at the github desktop... so far, only used git commands from within Cygwin64...
what tool are most of you guys using (on Windows) ?

digitalentity, 2016-10-18:
Is the problem still valid with 1.3-RC1?

digitalentity, 2016-11-06:
No feedback, assuming the issue is resolved. Closing.

closed 2016-11-06 by digitalentity