iNavFlight/inav #650: Clear indication of compass failure

krzysztofmatula, 2016-10-04:
A number of compass-related issues appear recently. In case it happens in the field, there is no clear indication to the pilot that some navigational features (including emergency RTH) may not work.

For GPS and BARO we may observe telemetry and OSD reports. But there's no such thing for MAG.

What would be the best indication method:
- re-using some telemetry fields (e.g. GPS related)?
- led-strip error indication?
- arm-block (sufficient only for boot-time failure)?
- all the above?

stronnag, 2016-10-04:
There already is, LTM (and MSP) report compass heading.

digitalentity, 2016-10-05:
I think @krzysztofmatula is talking about indicating that compass is not available (misdetected, failed).

krzysztofmatula, 2016-10-05:
Correct. 
The heading reports sent over telemetry do change (based on ACC/GYRO) even if MAG hardware fails or is not detected. It is quite hard to judge the HW state out of these readings.

nibborg, 2016-10-07:
Maybe the LCD screen can be used. At this moment is prety useless.
Something like Mag:No , Fail , Ok on the screen on the startup .
This is indeed usefull for the other sensors too (accel, gyro, sonar, gps...)

At this time exist the boot-time event logging, but you need to connect the pc to the board.
Some indications on the screen will be very helpfull.

stronnag, 2016-10-07:
It also exists via telemetry for mwp and probably ezgui. This is an OSD problem, not a iNav problem. iNav already publishes the sensor status on demand.

giacomo892, 2016-10-07:
@stronnag happens that the mag is reported as OK but when you have a look in the sensor tab mag value are stuck. 
Heading is calculated only with gyro with no indications

stronnag, 2016-10-08:
got it. Status is only established at boot. Quite impressive how accurate the IMU is if you disconnect the mag after boot.

giacomo892, 2016-10-08:
@stronnag yes :) but tend to drift if you move fast :)

closed 2016-11-03 by krzysztofmatula

krzysztofmatula, 2016-11-03:
Continued in: https://github.com/iNavFlight/inav/pull/744

PR #731 (2016-11-03): ledstrip: blue warning on compass failure
- added new generic ledstrip warning: HW_ERROR (BLUE-BLACK)
- HW_ERROR warning set on compass failure (on init or in runtime)

fixes #650.
PR #744 (2016-12-12): Hardware failure detection and warning
Continuation of #731. Fixes #650.

Added new generic ledstrip warning: HW_ERROR (BLUE-BLACK)
HW_ERROR warning set on compass failure (on init or in runtime)

Things to do:

- [x] ~~Create failure detection for all available sensors (GYRO, ACC, BARO, MAG, SONAR, GPS)~~
- [x] Make a way to indicate sensor failure for machines w/o ledstrip
- [x] Report sensor health via telemetry and MSP
- [x] ~~Configurator support for clearly indicating the failure~~
- [x] Block ARMing if one of configured sensor fails
- [x] Make sensor selection explicit (no autodetection [... cut]