iNavFlight/inav #655: Airplane takeoff assistant

digitalentity, 2016-10-04:
Proposed a new flight mode `TAKEOFF ASSIST` for fixed-wing aircrafts. See also: #577 #564

Logic:

1. Enable `TAKEOFF ASSIST` + `ANGLE` flight mode, ARM, bring throttle above idle, motor spin at `min_throttle` regardless of actual throttle stick position
2. Throw the airplane. When launch (throw or catapult) detects acceleration it will wait for pre-defined time and bring motors to takeoff throttle. Throttle stick is still ignored. During this and previous stage roll and pitch integrator will be forced to zero for stability. Roll setpoint is kept at zero, pitch setpoint is kept at pre-defined climb pitch.
3. During takeoff airplane will try to maintain zero roll, zero yaw rate and pre-defined climb pitch.
4. Takeoff is considered finished when pilot moves the roll/pitch stick.

I'd like to have this implemented as soon as possible for very selfish reasons :smile: 

To find thresholds for acceleration and time I'll be grateful if fixed-wing pilots will provide some blackbox logs with throw-launches.

oleost, 2016-10-04:
Great. https://drive.google.com/folderview?id=0B-yHzNtOXxbESEJFRzlYMGdoRk0

I gave lots of logs laying around, is at least a start. 

May i suggest make the takeoff assist mode compatible with every flight mode? 

So in theory one could actually have takeoff assist always enabled. And simple let it switch to whatever mode you have selected when you move the sticks. (be it acro, passthrough, RTH)

oleost, 2016-10-04:
If MOTOR_STOP is enabled - motors are not spinning at all (will spin up briefly to indicate activation).

In my opinion I would not have it different for motor stop. 

Motor stops is for stopping motors when throttle is cut, in this case we have throttle on transmitter and it's natural to keep props spinning whether motor_stop are enabled or not?

digitalentity, 2016-10-04:
That's intended for bungee laugh when prop should not be spinning for a
while after launch detected.

oleost, 2016-10-04:
I understand, maybe have it configurable with its own CLI command?

oleost, 2016-10-04:
Just me personally, I don't like to not have motor_stop on airplane. 

But on the other hand I think it's greatly beneficial to have the proper already spinning when your hand throw the plane.

sppnk, 2016-10-04:
@oleost, IMO, not for bungee launching At least for me is very pleasant and safe to bungee launch the plane, and one or two seconds later apply throttle. I am launching lately aided by a rubber cord, no ramp, my Mini Talon, wich have a big propeller in the tail.

oleost, 2016-10-04:
@digitalentity You need more logs from my plane or do you have enough?

Schalonsus, 2016-10-04:
If the weather clears up i can make some logs with bungee.
The other plane i throw has no logger/ram.

digitalentity, 2016-10-05:
@oleost a few more logs would make calculations more consistent. From your log:
1-st sample: >2g X-acceleration for 80ms, >2.9g acceleration for 48ms, pitch -18deg, roll -8 deg
2-nd sample: >2g X-acceleration for 80ms, >2.9g acceleration for 32ms, pitch -18deg, roll -5 deg

This is what I'm looking for:
[image]

I think just looking and X-axis (forward) acceleration and plane attitude (should be level enough, ) is enough. We just need more cases of successful launches to make a reliable threshold for launch.

oleost, 2016-10-05:
@digitalentity Il send you an mail with rest of logs.. Also one should probably have other planes aswell.

oleost, 2016-10-05:
@digitalentity If weather permits I will do a couple of "dry" hand launch without motor spinning.

digitalentity, 2016-10-05:
@oleost thanks! I'm already hacking some code together.

digitalentity, 2016-10-05:
Looks like in most cases X-axis acceleration is >1.9G for at least 40ms. From logs I see that such acceleration threshold is never breached prior to throwing. Combined with limiting maximum angle to 30 deg we should get a reliable launch detector.

It may fail in case of a week throw (in logs I have it's less than 10% cases). In future we can extend it with GPS speed threshold (if available) and watch for either of thresholds.

digitalentity, 2016-10-05:
Ok, I've hacked some code together. Completely untested and won't fit into F1 boards at the moment.

oleost, 2016-10-05:
@digitalentity great will test when I get the time. 

What did you end up doing regarding motor spinning and motor stop?

digitalentity, 2016-10-05:
My intention is to respect MOTOR_STOP. 
If MOTOR_STOP is enabled:

1. Before launch - motor stopped
2. Delay after launch - motor stopped
3. After that - launch throttle

If MOTOR_STOP is disabled:
1. Before launch - motor spinning at idle
2. Delay after launch - motor at idle
3. After that - launch throttle

This may not work as intended, I didn't have a chance to test anything yet. Please don't do any tests with props on.

Procedure:

1. `NAV LAUNCH` mode has to be selected before arming. It won't activate in Configurator but firmware will keep in mind that user wants to launch.
2. After arming `NAV LAUNCH` mode is activated and launch procedure is executed
3. Launch procedure is aborted and `NAV LAUNCH` mode is disabled if
 a. Pilot moves the right stick during launch
 b. Pilot disables the `NAV LAUNCH mode` explicitly

`NAV LAUNCH` mode can't be re-activated in flight, only at the moment of arming.

oleost, 2016-10-05:
@digitalentity Not agreeing with you on the respecting the motor_stop part, since as long as you stop the motors when throttle stick is zero you already are respecting the motor stop feature. 

What this mode needs a configurable command that is nav_fw_autolaunch_bungee_mode which has decide if the property should be spinning or not. But that's my opinion. 

On the other hand I got to do some testing today with prop off. 

When enabling NAV launch and arming it started to stabilize, I threw plane and it detected launch and started to spin motor. So that worked as expected. 

But it didn't detect input on right stick and therefor I was locked out of controlling the airplane until I manually turned of the NAV launch flight mode. 
Tried switching between passtrough / acro / horizon without that helping.  So there is an bug. 

Got blackbox and cli log I can provide later if you want.

digitalentity, 2016-10-05:
@oleost I didn't expect it to work flawlessly on the first try, thanks! Launch detection works, that's good. This addition is still way to go, but we are getting somewhere.

The most benificial part of using launch mode is that PID integrators are disabled until launch meaning the airplane won't have windup problems right after launch (this is the biggest problem when launching Ranger in ANGLE mode). Stabilisation from arming to motor start is done using only P and D.

After fixing a bug with sticks and sorting out motor stop - this should be ready.

digitalentity, 2016-10-05:
@oleost @Schalonsus @sppnk regarding motor-stop and bungee launch:
How about having two values of throttle for launch:
`nav_fw_pre_launch_thr` - throttle value applied before launch and right after launch until `nav_fw_launch_motor_delay` has elapsed.
`nav_fw_launch_thr` - throttle value to use during climb after launch detected and `nav_fw_launch_motor_delay` has elapsed.

digitalentity, 2016-10-05:
Ok, this is trickier than expected. Looks like MOTOR_STOP only respects raw stick position - there's currently no way of stopping the motors from the flight control logic. How about we keep it as-is initially and keep "bungee launch" improvement as a task?

digitalentity, 2016-10-06:
Let's continue the discussion in https://github.com/iNavFlight/inav/pull/664

closed 2016-10-06 by digitalentity