iNavFlight/inav #659: cant run inav on chrome

joelmatousek, 2016-10-04:
I was recently trying to run inav on my chrome, but when I run the file, it says "could not load manifest." 
How do I run  it, I am using chrome

digitalentity, 2016-10-05:
Unable to reproduce. INAV Configurator works fine. Possibly issues with your Google Chrome installation.

DzikuVx, 2016-10-05:
@joelmatousek also no problems here

Ralf-W, 2016-10-05:
Been there, done that: I downloaded the zip file from GITHUB and called it e.g. "inav-configurator-master_12.08.2016.zip". Unziping this archive results in a folder named "inav-configurator-master_12.08.2016" that contains a folder "inav-configurator-master". 
Opening "inav-configurator-master_12.08.2016" in Chrome results in "could not load manifest." 
Opening "inav-configurator-master" in Chrome results in successful installation of iNAV Configurtor.

joelmatousek, 2016-10-05:
@Ralf-W thanks! Tried it and it worked

DzikuVx, 2016-10-05:
OK, since solved, I'm closing

closed 2016-10-05 by DzikuVx