iNavFlight/inav #660: sp racing F3 external mag Problem

jereitor, 2016-10-04:
Hello, I have problems with HMC5883 in FC SP Racing F3 in Betafligth is solved. It will be solved in Inav?.
Thank you.

Sorry my bad English much.
Translated with google translate

digitalentity, 2016-10-05:
Please share details of your setup.

Recently we are encountering numerous issues with HMC5883 chips due to lower error tolerance for critical sensors. Cleanflight and Betaflight don't care, but for INAV compass is a critical component.

jereitor, 2016-10-05:
I am sorry
https://github.com/betaflight/betaflight/pull/626

DzikuVx, 2016-10-12:
I would have to say: let's close it. I'm using HMC5883l with my SPRF3 without any problems now.
@stronnag @digitalentity  Thoughts?

stronnag, 2016-10-12:
I agree, I've had no mag issues since @digitalentity fixed up the last set of driver issues.

closed 2016-10-12 by martinbudden