iNavFlight/inav #665: Flip32 Internal Baro not detected

DzikuVx, 2016-10-05:
Internal baro MS5611 is not detected on Flip32 full board in latest _development_ b3eff8d832591182db38526c7f825a98bab33713

Looks like we have more problems with I2C devices detection than reported MAGs

```
Time Evt Parameters
   0:  0 
   0:  1 
 500: 19  (0, 0, 0, 0)
 500: 19  (1, 1, 0, 2)
 500: 19  (2, 2, 0, 2)
 500: 19  (3, 3, 0, 2)
 500: 19  (4, 4, 0, 2)
 500: 19  (5, 5, 0, 2)
 500: 19  (6, 6, 0, 2)
 500: 18  (7, 6, 0, 1)
 500: 18  (8, 6, 0, 1)
 500: 18  (9, 6, 0, 1)
 500: 18  (10, 6, 0, 1)
 500:  2 
 601:  3 
1137:  9  (2, 0, 0, 0)
1253: 10  (3, 0, 0, 0)
1304: 11  (1, 0, 0, 0)
1304: 12  (2, 0, 0, 0)
2963: 13  (0, 0, 0, 0)
2963:  4 
3464:  8 
```

digitalentity, 2016-10-05:
@DzikuVx what's the compiler you use?

stronnag, 2016-10-05:
For me 6.2.0 and 4.9.3 both result in failed baro detection (as earlier discussion against a PR that I'll look up later when I have better access)

digitalentity, 2016-10-05:
I have a Flip32 somewhere around. Will test soon.

DzikuVx, 2016-10-05:
4.9.3
And it is not delay problem as far as I can say. Tries delay of 5s and it still failed

DzikuVx, 2016-10-05:
It happened somewhere after a5f7067d9cd2a196b2dda5009d2349dc77e49404

DzikuVx, 2016-10-05:
OK, I've found it
Last working commit is 28e262e4c448697342aec62a647a3435fd3b39ea
MS5611 detection stopped working with 9ff56d316c3b8646b9487562e947bd91929006fb

@digitalentity @martinbudden

martinbudden, 2016-10-05:
@DzikuVx , I pulled that over from Cleanflight and just assumed it was OK.

In retrospect I think it was a bad idea - that code is too aggressive in dealing with errors.

DzikuVx, 2016-10-05:
Let's wait for @digitalentity oppinion on it, but I vote for a revert

martinbudden, 2016-10-05:
Me too, although I think the `return !(state->error);` can remain.

digitalentity, 2016-10-05:
Please don't revert anything yet. I'm standing on a bus with logic analyser and baro doesn't return ACK with latest commit. Moreover with 28e262e my Flip32 doesn't boot at all.

There is something wrong with I2C driver.

DzikuVx, 2016-10-05:
My Flip32 is booting without problems

digitalentity, 2016-10-05:
Found the issue. Might cause compass mis-detections as well.

DzikuVx, 2016-10-05:
OK, sine #668 fixes this problem, I'm closing this issue. 
The only question is should we wait for 1.3 or create 1.2.3 ? @digitalentity ?

closed 2016-10-05 by DzikuVx

martinbudden, 2016-10-05:
@DzikuVx , @digitalentity Personally I think we should do a 1.3 now, with what we have.

I'm in favour of rapid release cycles. 1.3 doesn't have any real major features, but it is an improvement and it puts a line in the sand for further development.

stronnag, 2016-10-05:
I also favour a more rapid release cycle. However, I'd also welcome the opportunity to flight test fundamental fixes (vice just bench test) prior to release.

DzikuVx, 2016-10-05:
I would like to confirm that we solved Baro and Mag issues and then do 1.3
Perhaps with #658

digitalentity, 2016-10-05:
@martinbudden I agree. However I think we can merge a few more things in 1.3: at least #644 #658

digitalentity, 2016-10-05:
Let's start moving issues to 1.4 milestone.

martinbudden, 2016-10-05:
Yes, by all means add in one or two other PRs. But we should avoid the trap of "I'll just add one more..."

I also think we should allow a bit for time for the release candidates, so that people like @stronnag can do testing before the full release. So at least a week between 1.3-RC1 and the 1.3 release, probably longer.

digitalentity, 2016-10-05:
Here's the interesting question - we decided to do releases from master, but what about RCs?

digitalentity, 2016-10-05:
I've transfered quite a lot to 1.4 milestone. Please review.

martinbudden, 2016-10-05:
RCs should also be done from master.

Procedure is:

1. Merge development code into master.
2. Issue RC
3. Once RC is issued only bug fixes on master, other work on development
4. In major issue detected on RC, it is fixed and new RC issued
5. When RCs have been sufficiently we issue a full release
6. There may be several RCs (betaflight went up to RC14 for version 3.0, but that is a bit extreme). But two or three RCs should be fairly normal.

digitalentity, 2016-10-05:
3. Bugfixes directly to `master`, then merge from `master` to `dev` as always?

martinbudden, 2016-10-05:
Yes.