iNavFlight/inav #666: MSP RX input doesn't work

anbello, 2016-10-05:
If I choose RX_MSP as Receiver Mode on Configuration tab then on Receiver tab I have no channels, "Control sticks" have no effect and if I use MSP program to send RC commands (EZ-GUI on Android or PyMultiWii on PC) they have no effect.
Sometime ago I did it with cleanflight and it worked, maybe with iNav I have to do some other setting in other place.

anbello, 2016-10-07:
Maybe no one answered because of the number of this issue :)

oleost, 2016-10-07:
Or if you provided more information? 

What board, what iNav version, cli dump, verify that the same method your using now are still working in Cleanflight.

anbello, 2016-10-07:
Sorry

I have the same problem with two quad:
1) naze32 10dof rev6 (clone from banggood) with iNav 1.2
2) SP Racing F3 brushed with iNav 1.2.1

for both I have an esp8266 wifi serial bridge connected to uart1 that works with ez-gui for telemetry but not for control (as I said before)

for 2) I tried the same method with last cleanflight and it worked

When I return to home from my work I will post some dump

oleost, 2016-10-07:
Can you try iNav 1.1 as well to see if it works there?

martinbudden, 2016-10-07:
It is possible that some of my receiver changes broke this. Unfortunately I won't be able to look at this until late next week.

anbello, 2016-10-07:
In this week end I can try with iNav 1.1 and I will report here.

digitalentity, 2016-10-07:
If this is a bug, it should go into 1.3

anbello, 2016-10-11:
What I said before is partially incorrect.
On the SP Racing F3 brushed with iNav 1.2.1 if I set RX_MSP as Receiver Mode it works.
On the Naze32 rev6 with iNav 1.2 and 1.2.1 it doesn't works (as I said above). If I re-flash iNav 1.1 I have the correct behaviour and RX_MSP as Receiver Mode  works.

stronnag, 2016-10-11:
That is because RX_MSP is only enabled for targets with > 128K flash (see src/main/target/common.h)

This is a feature.

martinbudden, 2016-10-11:
RX_MSP was removed from the sub 128K targets as a ROM saving measure. So this is deliberate and not a bug. Closing issue.

closed 2016-10-11 by martinbudden

anbello, 2016-10-11:
Thanks and sorry for having wasted your time, I did not know that this feature had been removed.

martinbudden, 2016-10-11:
@anbello - no worries. But reports are always welcome.

digitalentity, 2016-10-12:
@anbello, @martinbudden wiki updated to reflect lack of MSP RX support on NAZE & CC3D
https://github.com/iNavFlight/inav/wiki/Hardware-and-feature-support-map