iNavFlight/inav #670: How do you wire pwm receiver to naze32 running inav

joelmatousek, 2016-10-06:
I am new to inav, and I'm having trouble setting it up. I'm using pwm for my reciever, and I don't know how to wire it up properly any help

joelmatousek, 2016-10-06:
I'm also using a gps

DzikuVx, 2016-10-07:
You will have plenty tips here: https://www.google.de/search?client=opera&q=naze32+pwm+setup

andre92, 2016-10-07:
When using uart 2 with pwm receiver, you lose the last 2 channels and channel 3-6 shift right 2 pins. Wire up the gps first to pins RC3 and RC4 and make sure it works (configure it on uart2 in ports tab), then connect the first 2 pwm channels as normal, and the next 4 on RC5-8. Use the receiver tab to verify all channels are connected to the right pin. 
REMOVE PROPS in case it accidentally arms.

jstremmler, 2016-10-07:
Use a ppm encoder, which you can buy for $ 4, configure your GPS to UART2 and you will be happy!

Here you can buy the ppm encoder: [https://www.aliexpress.com/item/8-Channel-PPM-Encoder-Module-V1-6-For-Pixhawk-PPZ-MK-MWC-Flight-Controller-66/32545897287.html?spm=2114.13010608.0.0.cv4eKo](url)

digitalentity, 2016-10-09:
@joelmatousek please consider using forums for support.

As for your issue - I recommend using a PPM encoder.

commit 8c505084 (2015-03-25): Update telemetry documentation now that telemetry_provider is gone.

Closes #670.
closed 2016-10-09 by digitalentity