iNavFlight/inav #671: Overwrite blackbox memory when it is full

sppnk, 2016-10-07:
Many of us dont use blackbox logs a lot, or just when having problems. It would be nice to have a parameter "blackbox_overwite = 1/0" in order to have the memory overwritten from begining each time it is full. 

This way, we dont have to erase it at launch time, (what is annoying and I usually dont do) and the data of the last flight would be always available.

flounderscore, 2016-10-07:
I've asked that question before and the answer I got was that it's not possible because you can't overwrite the flash. It has to be zeroed/erased first, which is slow.

andre92, 2016-10-07:
@flounderscore so it is not as easy as resetting the flash write address to zero?
I would be really interested in this feature. It eliminates the need to manually erase the flash, plus there will always data of the last minute of in flight problems

digitalentity, 2016-10-07:
Indeed, flash chips we use are slow when doing erase operations - erasing the full chip takes up to 30 seconds.

oleost, 2016-10-07:
Just a tip,  turn down the logging rate to minimum, should last many flights. And usually its enough unless you are tuning pids

stronnag, 2016-10-07:
And the interleaved header / data in the on chip logs is not conducive to treating flash as a ring buffer, even if overwrite was electrically possible.

andre92, 2016-10-07:
@digitalentity is it possible to erase a single page? That way we can leave the header intact. And if so, how much cpu time would it take to erase one page?

sppnk, 2016-10-07:
@stronnag, I am not sure to understand you. What I would like is that every time I launch the aircraft the blackbox begins to be overwriten from zero, just to have an entire flight recorded.

@flounderscore , I should read the datasheet of the memory, but, is it really neccesary to erase before write it?

stronnag, 2016-10-07:
And if I fly my quad 6 times in an afternoon, I'd like all the logs, not just the last one.

digitalentity, 2016-10-09:
There are very few targets with onboard memory. Most targers must use OpenLog (virtually unlimited storage). Some modern FCs have SD card slot onboard giving you virtually unlimited storage as well. I don't think it's worthwile investing effort in supporting something that will only be used in 1% cases. 

For those who need to record long flights to onboard flash chip I recommend reducing loggint rate to 1/32 - the log will still be usable to replay a mission or troubleshoot GPS issues. The only thing that requires 1/2 or 1/1 logging rate is PID tuning.

sppnk, 2016-10-10:
Personally, I only use blackbox when trying to fix issues. I purchased an openlog for one of my naze FCs and it worked fine solving a problem. In CC3D I have 2Mb,  I guess. 

But when I only go to fly and enjoy, I dont care about blackbox, althought I miss it if I have an issue.

I use 1/32 logging rate, and was talking about to have a parameter to switch on/off this.

Never mind and thanks. Closing this...

closed 2016-10-10 by sppnk