iNavFlight/inav #675: Airplane throttle pitch compensation

digitalentity, 2016-10-09:
Airplanes need extra throttle when climbing and lower throttle when diving. We have a `nav_fw_pitch2thr` parameter that is active in navigation modes.

We also have a very cool throttle `throttle_tilt_comp_str` parameter for copters that can be active also in non-navigation modes and makes flying with manual throttle control considerably easier by adding extra throttle when drone is not level. It's based on physics and trigonometry and calculates a vertical component of thrust.

Unfortunately `nav_fw_pitch2thr` is a dumb multiplier for pitch angle which is totally incorrect. We need some math that will be more connected to real world.

Ideas anyone?

oleost, 2016-10-09:
Do we need anything more? ( I got no other suggestions )

Expect maybe making the curve non-linear if its not already is?

[image]

digitalentity, 2016-10-09:
Let's also assume that trust/weight ratio is 1.5 meaning that airplane is able to hold it's own weight at 67% throttle.
Let's also assume out cruise throttle is 40%, this means that propeller at 40% generates enough thrust to equalise the drag. We can roughly assume that the cruise thrust is 0.6 of airplane weight (this is actually not true due to effects of airspeed but let's drop that part for now).

**Level flight:**
If flying level horizontal component of weight is zero, no extra throttle is needed.

**Diving**
When diving, airplane weight will assist in keeping velocity, so we need to reduce throttle according to thrust/weight ratio. With our thrust/weight ratio we'll be reducing throttle by 67% * sin(pitch) meaning we should be flying at zero throttle at 38 deg dive.

**Climbing**
Same thing as diving. Full throttle will be reached at 63 deg climb.

The above model is still flawed because propeller thrust varies with airspeed but without airspeed sensor we can't really do much. Simply speaking trust/weight ratio at speed will be smaller than in static test.

Basically we need to define rough thrust/weight ratio (at cruise speed, we'll have to guess it :smile:) and act on it to compensate for pitch in a smarter way.

closed 2018-05-27 by stale[bot]