iNavFlight/inav #684: Airplane PID attenuation

digitalentity, 2016-10-12:
On airplanes output of PID controller must be scaled according to TAS (true air speed) - at higher speeds control surfaces are more effective and therefore less throw required to achieve desired response. 

Not all aircrafts are equipped by airspeed sensors (pitot meter), also INAV doesn't support pitot sensors yet.

In general one should tune the airplane to fly good at cruise throttle. Control surface response should be **increased** when throttle is below cruise and decreased otherwise. 

My suggestion is to change TPA logic by one that is done in ArduPlane:
```
float Plane::get_speed_scaler(void)
{
    float aspeed, speed_scaler;
    if (ahrs.airspeed_estimate(&aspeed)) {
        if (aspeed > auto_state.highest_airspeed) {
            auto_state.highest_airspeed = aspeed;
        }
        if (aspeed > 0.0001f) {
            speed_scaler = g.scaling_speed / aspeed;
        } else {
            speed_scaler = 2.0;
        }
        speed_scaler = constrain_float(speed_scaler, 0.5f, 2.0f);
    } else {
        if (channel_throttle->get_servo_out() > 0) {
            speed_scaler = 0.5f + ((float)THROTTLE_CRUISE / channel_throttle->get_servo_out() / 2.0f);                 // First order taylor expansion of square root
            // Should maybe be to the 2/7 power, but we aren't going to implement that...
        }else{
            speed_scaler = 1.67f;
        }
        // This case is constrained tighter as we don't have real speed info
        speed_scaler = constrain_float(speed_scaler, 0.6f, 1.67f);
    }
    return speed_scaler;
}
```

They don't have TPA attenuation value, just cruise throttle (TPA breakpoint).

For TPA breakpoint of 1500 and TPA attenuation of 0.75 this will yield the following:
[image]

As you can see - PIDs are 67% stronger at zero throttle, same at cruise throttle and attenuated roughly the same way up to full throttle.

Benefit of this approach is having stronger stabilisation at low speeds (when coming in for a landing or taking off with stopped motors).

Please share your thoughts

martinbudden, 2016-10-12:
@digitalentity wouldn't a TPA breakpoint of 1200 and a TPA attenuation of about 0.45 give something resembling the ArduPlane code at low throttle?

digitalentity, 2016-10-13:
In ArduPlane throttle/pid attenuation is non-linear, control is reduced very fast at low throttle and much slower at high throttle. You might find good TPA to have decent flight characteristics at cruise throttle and above, but you'll have to sacrifice neer-stall stabilisation.

digitalentity, 2016-10-13:
I might be able to flight-test the new fixed-wing TPA today and see if it improves the flight characteristics.

digitalentity, 2016-10-13:
Let's continue the discussion in #685

closed 2016-10-13 by digitalentity