iNavFlight/inav #694: Regraded tricopter performance with #397

stronnag, 2016-10-14:
Firstly, apologies for not catching this earlier. 

I flew my tricopter (again) with  d54a6c5. I noticed both visually and audible that the motors were significantly less smooth and there were visually obvious vibrations, pitch in particular. Switching back to  efd8885 returned performance to normal (the vehicle is somewhat noisy anywhere, but with #397 noticeably more so). I have not seen this variation in smoothness with non-servo craft (e.g. quads) for #397. My βF mini-tri does not exhibit the same degradation in performance with unsynched motor update; otoh the mini / βF is PWM, vice OS125 on the iNav machine. Tomorrow (if it's not raining) I will test the iNav machine with PWM.

Some logs at http://seyrsnys.myzen.co.uk/inav_1.3-397/
* LOG00001.TXT -  d54a6c5 (i.e. with #397), please ignore the obvious mag failure; this is cabling fault.
* LOG0000{6,7}.TXT - efd8885 (i.e. without #397)
* tri_evo_cli-dump.txt - cli dump (for d54a6c5, but the only difference is FEATURE v. set motor_pwm_protocol )

digitalentity, 2016-10-14:
@stronnag please try with
set motor_pwm_rate = 1000 or 2000

Your test with #397 didn't actually increase motor update rate compared to
ordinary pwm, but merely switched the motor update protocol.

stronnag, 2016-10-15:
Thanks, I'll retest as soon as possible.

stronnag, 2016-10-15:
Works fine with 1k and 2k. In my (admitted weak) defence, the documentation was incorrect. See PR #698 for an update.

Closing as (a) user / documentation error and (b) documentation PR #698

closed 2016-10-15 by stronnag

digitalentity, 2016-10-16:
Great! Thanks for updating the docs.