iNavFlight/inav #695: Naze32 Uart1 Problem

jankanapka, 2016-10-14:
I have a problem concerning uart1. It works perfectly when I connect it to a PC via built in usb connector. However, when I try to connect it with FTDI adapter or OSD it won't work. Is it the hardware or soft problem? I am running the Inav 1.2.1 stable directly from configurator. Could you help me please?

andre92, 2016-10-15:
It is a hardware limitation on the Naze, since the usb is connected to the uart, and uart can only be used for one thing at a time. When OSD is connected and you power it, USB will stop working. If you need to configure the OSD with your computer, you need to disconnect it from the Naze.

jankanapka, 2016-10-15:
I would like to point out that I never run two things simultanously. My problem is that I cannot connect to UART1 via tx/rx pins at all, when usb is disconnected.

jankanapka, 2016-10-17:
Has anyone experienced similar thing?

oleost, 2016-10-17:
Tried flashing cleanflight? If it does not work then it's probably not iNav fault and you are doing something wrong.

digitalentity, 2016-11-21:
Most likely a hardware issue.

closed 2016-11-21 by digitalentity