iNavFlight/inav #702: linker type mismatch warnings for pca9685setPWMFreq

stronnag, 2016-10-16:
The gcc 6 linker emits the following:
````
./src/main/drivers/io_pca9685.h:4:6: warning: type of 'pca9685setPWMFreq' does not match original declaration [-Wlto-type-mismatch]
 void pca9685setPWMFreq(float freq);
      ^
./src/main/drivers/io_pca9685.c:86:6: note: type mismatch in parameter 1
 void pca9685setPWMFreq(uint16_t freq) {
      ^
/usr/arm-none-eabi/include/sys/_stdint.h:36:20: note: type 'uint16_t' should match type 'float'
 typedef __uint16_t uint16_t ;
                    ^
./src/main/drivers/io_pca9685.c:86:6: note: 'pca9685setPWMFreq' was previously declared here
 void pca9685setPWMFreq(uint16_t freq) {
      ^
./src/main/drivers/io_pca9685.c:86:6: note: code may be misoptimized unless -fno-strict-aliasing is used
````
Due to the reference in `src/main/io/pwmdriver_i2c.c` this would benefit from developer review.

DzikuVx, 2016-10-24:
@stronnag which gcc ?

stronnag, 2016-10-24:
gcc 6.2. 
./src/main/drivers/io_pca9685.c will compile OK, because it does not reference the header files, just a forward reference to pca9685setPWMFreq().

src/main/io/pwmdriver_i2c.c includes ./src/main/drivers/io_pca9685.h, and thus also compiles its function prototype OK. 

So static compilation alone cannot find the bug; however the clever gcc 6.x link time checking finds it.

martinbudden, 2016-10-25:
@DzikuVx , when you are fixing this can you also remove the `#include config/config.h"` and 
`#include "fc/runtime_config.h"` from `drivers/io_pca9685.c`? They are not required and also violate the principle that drivers should not call or include "upward".

DzikuVx, 2016-11-01:
Moved to #736

closed 2016-11-01 by DzikuVx