iNavFlight/inav #704: Does RTH work with only GPS but without compass?

kejunxia, 2016-10-16:
Sorry this is a question instead of issue not sure where else to ask about it.

I am new to quad copter. And I ordered a quad comes with a flight controller without a compass. I then ordered a GPS module which doesn't come with a compass from here for CC3D/SP F3 controller. Is it still possible to get RTH working maybe headless mode? 

http://www.banggood.com/Mini-Ublox-M8N-GPS-Module-NEO-M8N-GPS-for-APM-2_52_62_8-CC3D-SP-Racing-F3-Naze32-Flip32-PX4-p-1035454.html?rmmds=search

stronnag, 2016-10-16:
Please ask support questions on RC Groups, IRC or gitter. To answer you question, for a multirotor, you need a compass.

Please now close this issue.

closed 2016-10-16 by DzikuVx

kejunxia, 2016-10-16:
Thanks for the quick action. I will ask there then. @DzikuVx Really appreciated if you could give me a quick guidance of the info you mentioned in this issue https://github.com/iNavFlight/inav/issues/660. 

- Were you referring HMC5883l on your SPRF3 or you added an external compass on to it? Is it possible to add [this compass](http://www.banggood.com/Electronic-Compass-Module-Three-Axis-Magneto-Resistive-Sensor-p-926151.html?rmmds=detail-left-viewhistory) on a normal SPRF3
- By the way, is barometer necessary as well?