iNavFlight/inav #710: Wrong RTH trajectory and throttle

favn1585, 2016-10-18:
Hello! First I wanna said thanks for developers. Guys, you are awesome! Your flight core works much better than cleanflight's one.

Now my questions.

1. I have Zeta fx-61 with Emax Skyline 32 Advanced + OSD. It has 25p16vp magnetometer on the board. I use latest (1.2.1 NAZE) firmware (did not try 1.3.0 yet). Gps is ublox neo-6m. I have board rotation to 90 degrees. I set 0 (integral) PID values for PITCH and ROLL, I change them after my servos start mowing after increasing throttle in angle mode, probably this is because this PID values more for muiltirotors. 

[image]

Main problem is when I turning on RTH, my plane has such trajectory while flying home. I think that there may be some delay with mag and while plane correcting it's trajectory, his heading already changed.

[image]

I was trying to turn off mag and use gps heading (set mag_hardware = 1) but due to issue with that (https://github.com/iNavFlight/inav/issues/620#issuecomment-248415157) I decide that it's not good idea, and why I need to turn it off if my mag. is working fine when I connect it to configurator. I see that magnetometer is working because flying wing returning to home even if it's 500-800 m away, but unfortunately it flies with zigzag trajectory.

2. When I completely turn off my radio, my flying wing stop spinning motor (throttle 0) and it's jut gliding to RTH position. I expect it to fly to home position (using mag/baro to increase throttle). I set:

set nav_rth_alt_mode = FIXED
set nav_rth_altitude = 10000 (100m)

Maybe I should increase "Throttle value while landing"? I have value 1000 there. 

Thank you in advance!

oleost, 2016-10-19:
1. Try and turn of mag.
1.1 Have you followed this guide? https://github.com/iNavFlight/inav/wiki/3.-Fixed-wing-guide#step-4-replace-default-pids 
2. https://github.com/iNavFlight/inav/wiki/9.-Failsafe
2.1 I suspect you dont have no pulse, or have set up your failsafe wrong.

Also for showing us your setup, use an CLI dump instead of "backup" function. And a blackbox log is also often helpful.

And please close this issue and use [rcgroups](http://www.rcgroups.com/forums/showthread.php?t=2495732&page=1000) thread of user support.

favn1585, 2016-10-19:
ok. thank you

closed 2016-10-19 by favn1585