iNavFlight/inav #712: LTM telemetry : add downlink quality check

sppnk, 2016-10-19:
I am lately testing LTM downlink telemetry and having a lot of problems, probably because of my radiolink (openlrsng), I was thinking if could be added a packet counter in the protocol, in order to check at ground if all the packets are arriving or not.

This way we could have another security and information for the radiolink.

This probably could be done inside the 'X' frame, a cyclic counter with one of the spare 4 bytes. What do you think?

DzikuVx, 2017-01-31:
I suppose we could use 1 byte for counter here. 0-255 and again. Putting in 1.7

sppnk, 2017-03-24:
Hi @DzikuVx , I am re-thinking about this, as I am playing with my LTM OLED receiver. 

I guess now that we dont need this, as the packets/second count can be done at ground side. Then a link quality check is easy, just checking if this value is less than a constant (depending on bps of the link).

What do you think? Closing this? Thks.

DzikuVx, 2017-03-24:
Let's keep it open since this is quite a useful idea. But I will remove it from 1.7

digitalentity, 2017-03-25:
LTM in general can be variable baud rate - there is no strict requirement about how fast packets should be sent out, so I agree, let's keep this open for now.

stronnag, 2017-03-25:
We have an extant Issue / PR with sample code to lower the rate / volume, so this is much better done in the ground station. The FC has no idea of what may be lost over the down link.

digitalentity, 2017-03-25:
This issue is about giving GCS a better hint to detect possible packet loss rather than "packets per second". As @stronnag pointed out LTM could be slowed down and pps value will be lower.

stronnag, 2017-03-25:
Really we are addressing the symptom rather than the cause. A better approach would be to offer lower rates (as air speed directly affects range), via the techniques explored in the code sample in https://github.com/iNavFlight/inav/pull/184. This could be implemented without the intrusive baud rate changes as air speed != serial speed for some telemetry devices, and the devices have memory for buffering. So:

* A CLI variable `set ltm_update_rate = [NORMAL/MEDIUM/SLOW]`
* A set of rate tables following the code example in https://github.com/iNavFlight/inav/pull/184

In future, if the baud rate implementation is changed, then this could become automatic on baud rate rather than set by the CLI variable and would allow airspeed equivalents of 2400 and 1200 baud for 3DR at least.

I will look to generating a PR along these lines.

stronnag, 2017-03-25:
There is now an (untested at the moment branch) ltm-rate-options

By my calculations, the rates are (bytes / second)

ltm_normal_schedule 203
ltm_medium_schedule 112
ltm_slow_schedule 73

We could reduce the slow rate by removing a G_FRAME (so there is one G_FRAME /sec), this would reduce the slow schedule rate to 59 bytes / sec.

I will at least do some bench testing tomorrow to confirm the theory, the above rates calculated by a ruby script that parses the source file; I guess that could be usefully dropped into 'support'?

stronnag, 2017-03-25:
Corrected rates, I forgot the protocol overhead:

ltm_normal_schedule 303
ltm_medium_schedule 164
ltm_slow_schedule 105 (removing one G-FRAME => 91 bytes /sec)

That matches extant observations

stronnag, 2017-03-25:
Bench tested, reported rates in mwp agree with the theoretical rates above. PR to follow flight tests (and any opinions on whether slow should report 2 or 1 G_FRAMES / sec). 

May be useful for long range stuff and particularly 'planes at low air speed telemetry rates.

digitalentity, 2017-03-26:
@stronnag looking forward to have that PR in, thanks!

stronnag, 2018-04-04:
LTM has the (% 256) frame counter for some time now, closing.

closed 2018-04-04 by stronnag