iNavFlight/inav #713: Mag detection on NAZE32

jstremmler, 2016-10-20:
iNav 1.3.1.- RC1 creates the same problems when detecting the  magnetometer like version iNav1.2.0 did before. 

Magnetometer on my NAZE32,  a BMP 280, is not found anymore.

digitalentity, 2016-10-20:
BMP280 is a barometer. What sensor has issues - magnetometer or barometer?

jstremmler, 2016-10-20:
Sorry, mixed it up ! 
 
- barometer = BMP280 (barometer on board)
- magnetometer = HMC5883 (magnetometer is on the board as well)
- CLI status command shows a huge number of i2C errors.

It is a NAZE32 Rev6. 10DOF from Bangood. With iNav version 1.2.1 everything works fine. And this time - when I compile the iNav development version 1.3.0-RC1 of iNAV with cygwin64 myself -  it shows the same behavior like the downloaded NAZE.HEX file from github. No magnetometer is detected with iNav 1.3.0-RC1.

digitalentity, 2016-10-20:
Ok. Please share the bootlog output

jstremmler, 2016-10-20:
Ok here is status and bootlog for iNAV 1.2.1 and for iNav 1.3.0 for the same NAZE32 Rev.6 10 DOF FC:

[image]
[image]

digitalentity, 2016-10-21:
Hm... interesting. NAZE Rev5 has no issues. @all, can somebody else confirm the issue on NAZE Rev6 ?

ServError, 2016-10-21:
I just dug up and flashed a real Rev6 and can verify his output.

jstremmler, 2016-10-21:
I also can confirm that Rev.5 works with iNav 1.3.0-RC1!

sppnk, 2016-10-21:
I have flashed right now a NAZE with 1.3.0.RC-1 and I have a enormous number of I2C errors as well.

digitalentity, 2016-10-23:
Ok, looks like we need someone who is in posession of Naze rev6 and a signal analyzer to get a grip what's happening on i2c bus. Apparently it's another i2c-related issue.

digitalentity, 2016-10-23:
I don't what this to hold off the 1.3 release though, we can release 1.3 with a known bug.

jstremmler, 2016-10-23:
DE, let me cite you from RCGroups: "Maybe we should stop support for the NAZE32" .

As the NAZE32 is a little outdated and causes a lot of problems (does not matter whether it is in Betaflight, Cleanflight or iNav; have look here : https://www.youtube.com/watch?v=czjtg-iXjxE9)

maybe this is the moment when iNav development should take the lead and officially skip any support for the NAZE32. 

What do you think?

martinbudden, 2016-10-23:
@digitalentity , I agree - we can release 1.3 with this as a known bug.

@jstremmler , I'm against dropping support for NAZE32. Reasons include:

1. While it's no longer advisable to buy a NAZE32, there are plenty of existing aircraft out there which have NAZE32s in them
2. Many of the brushed flight controller boards are based on NAZE32 (MicroSciSky, for example). Although increasingly brushed boards are F3 base.
3. Bugs found with the NAZE32 may indicate underlying bugs in the code, not necessarily NAZE32 specific problems.
4. The discipline of writing code for a smaller target has a tendency to improve the overall quality of the code.

digitalentity, 2016-10-23:
@jstremmler I don't think dropping any target is a good idea unless absolutely necessary. @martinbudden gave a very good list of reasons why we sould keep supporting this target.

As for this bug, it's likely still related to I2C driver. Something we do when initializing i2c bus causes some i2c device to go haywire. I suspect the bmp280 barometer device. 

Do we have any other F1/F4 target with bmp280 except NAZE Rev6? I suspect not, at least not many and not widely used.

jstremmler, 2016-10-23:
Tried my NAZE Rev6. 10DOF with Betaflight 3.0.1 (latest version). It shows same behavior! The magnetometer is not detected....

oleost, 2016-10-23:
Did you mean F1 or F4 or F1 to F4? 

My F3 spracingf3 mini is BMP280,  but that is an F3

oleost, 2016-10-23:
@digitalentity Did you mean F1 or F4 or F1 to F4?

My F3 spracingf3 mini is BMP280, but that is an F3

skaman82, 2016-10-23:
@jstremmler I think the additional sensors are disabled in betaflight on default. Try to activate the manometer in the config tab in the betaflight configurator or in the CLI.

digitalentity, 2016-10-23:
@oleost, I ment F1 **and** F4 - they share the same I2C bus code. F3 has different I2C implementation.

digitalentity, 2016-10-23:
BTW, @jstremmler can you test with 3 more configurations:
1. Both mag and baro disabled
```
set mag_hardware=1
set baro_hardware=1
```

2. Baro auto, mag disabled
```
set mag_hardware=1
set baro_hardware=0
```

3. Baro disabled, mag auto
```
set mag_hardware=0
set baro_hardware=1
```

I'm curious if this is related to a specific chip

jstremmler, 2016-10-23:
@skaman82 I am aware of that and tried to switch everything on in betaflight, but magnetometer did nor work.

And here are the results of the tests:
1. set baro_hardware=1 and set mag_hardware=1 works, no I2C errors but as both are switched off I don't have them and cantuse them.
2. set mag_hardware=1 and set baro_hardware=0 or =4 switches the barometer on but shows huge amount of i2C errors
3. set mag_hardware=0 or =2 and set baro_hardware=1 switches the magnetometer on, the barometer off and shows zero !!! I2C errors
4.  set mag_hardware=0 or 2 and set baro_hardware=0 or =4  switches the magnetometer to off , the barometer to on and shows a huge amount of I2C errors. 

For me this looks like switching the barometer to on is causing the problem....

digitalentity, 2016-10-23:
Hm... looks like a bmp280 issue. I think I have a bmp280 breakout board I can connect to a Naze Rev5 and test.

digitalentity, 2016-10-24:
Issue confirmed using bmp280 breakout board. Fix pending.

digitalentity, 2016-10-24:
Good news everybody, issue tracked down to I2C transfer timeout. BMP280 driver reads factory calibration data in one i2c transaction (24 bytes) which is taking too long and I2C driver hits the timeout. 

A dirty fix is simple increase the timeout. A proper fix would be to calculate timeout according to transation length and bus speed. We should also probably migrate to `micros()` instead of CPU-cycles there.

See #720

digitalentity, 2016-10-24:
TBH, the bug was always there, we inherited it from Baseflight which also suffers from it.
However everywhere a timeout is treated as a stuck bus and unstick procedure is started. Fortunately it's only one byte that gets stuck.

It's the recent change to I2C driver (https://github.com/iNavFlight/inav/pull/668) that loosened consitions for bus reset and revealed the problem in all of its ugliness.

digitalentity, 2016-10-24:
[image]

digitalentity, 2016-10-24:
PR #720 fixes this. Closing.

closed 2016-10-24 by digitalentity

digitalentity, 2016-10-25:
Is that a NAZE rev6 as well?

juzhen698, 2016-11-26:
Entering CLI Mode, type 'exit' to return, or 'help'

# STATUS
System Uptime: 45893 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 0.02
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=HMC5883
Cycle Time: 2008, I2C Errors: 10, config size: 1348

# VERSION
# INAV/NAZE 1.4.0 Nov 21 2016 / 12:34:05 (3823e46)

nipunnawanjana1, 2018-02-20:
it work for me https://youtu.be/9Es1NzxLyWE