iNavFlight/inav #715: Motors start non-idle on arm (SPF3 and EVO)

krzysztofmatula, 2016-10-20:
I observe that on first arm after powering up the motors start at high RPM which almost flips the quad if no RC control in immediately applied.

See this blackbox log - two motors start at >half throttle immediately after arming:
[LOG00008.TXT](https://github.com/iNavFlight/inav/files/543002/LOG00008.TXT)

Then, for subsequent arm actions the motors start idle as expected:
[LOG00005.TXT](https://github.com/iNavFlight/inav/files/543010/LOG00005.TXT)

Sometimes motors also start on other values, e.g. 1/4 (not critical, but it should also not happen):
[LOG00007.TXT](https://github.com/iNavFlight/inav/files/543012/LOG00007.TXT)

The attached logs are from inav1.3-rc1 and SPRACINGF3-EVO (600mm quad), but I'm observing this since some time (I think since inav-1.2) also on other machine based on SPRACINGF3 (250mm quad).

My configuration is similar to the default one:
```
# dump

# version
# INAV/SPRACINGF3EVO 1.3.0 Oct 20 2016 / 17:53:26 (613bf99)
# pflags
# Persistent config flags: 0x00000001
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -UNUSED_1
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -UNUSED_2
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature RX_PPM
feature VBAT
feature GPS
feature FAILSAFE
feature TELEMETRY
feature RSSI_ADC
feature LED_STRIP
feature BLACKBOX
feature CHANNEL_FORWARDING

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
[... 388 lines cut ...]
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 40
set yaw_rate = 40
set tpa_rate = 0
set tpa_breakpoint = 1500

# 
```

krzysztofmatula, 2016-10-24:
Attaching yet another log (indoors, no props, arming by stick command):
[LOG00009.TXT](https://github.com/iNavFlight/inav/files/549027/LOG00009.TXT)

The motors start at: 1168, **1551**, **1566**, 1150.
The rcCommand is 1150 (== min_throttle).

Am I the only one that experience this issue?
I think it is quite critical.

The settings are as attached earlier. Mainly defaults. Angle mode.

DzikuVx, 2016-10-24:
Seems like quad wants to turn because of mag hold. Could you Try arming without mag to confirm?

krzysztofmatula, 2016-10-24:
That's it!
[LOG00018.TXT](https://github.com/iNavFlight/inav/files/549115/LOG00018.TXT)

Arming with ANGLE (and no MAG) is also OK.
Then, enabling MAG makes the spike on motors for a fraction of second (bad!)

When arming with ANGLE AND MAG two motors spin constantly at >1500 with no throttle input (very bad!)

nomeda, 2016-10-25:
Hello, 
I have the same problems with Inav 1.2 on a Naze 32, if I arm with Mag on the motors spin up, but this happens only on the first arm. Unfortunately i do not have a black box, so I can not provide a log file, but the behavior is exactly the same as krzysztofmatula reports.

oleost, 2016-10-25:
@digitalentity Sounds like a critical bug, but is already present in iNav 1.2.

So in my opinion not neccesary for a 1.3 release, but should be mentioned as known bugs.

DzikuVx, 2016-10-25:
Another question, how far is mag away from any power line, in cm please 😁
It's just to make sure of something

krzysztofmatula, 2016-10-25:
It is about 5cm from power lines.
The test was done with no props. Idle motors don't take much amps.
The MAG readings in blackbox seem quite stable.
The issue manifests since nav-1.2. Didn't manifest earlier.
I bet for software issue.

DzikuVx, 2016-10-26:
I think I know what is happening here and probably there are two cases:

1. arming does not reset MAG_HOLD controller
2. enabling MAGHOLD does not reset MAG_HOLD controller

Also, is it possible that first you power the UAV and then move it to a place where you arm it (change orientation as well)?

krzysztofmatula, 2016-10-26:
> Also, is it possible that first you power the UAV and then move it to a place where you arm it (change orientation as well)? 

No, this is not my case.
Also the absolute orientation doesn't matter. I mean, I observe the issue when arming facing north, as well as south.

DzikuVx, 2016-10-26:
@krzysztofmatula take a look at #726 . I'm shooting blind but that should help it

krzysztofmatula, 2016-10-26:
Thanks! I'm going to test it today.

DzikuVx, 2016-11-01:
PR fixing issue in place, closing this one

closed 2016-11-01 by DzikuVx

krzysztofmatula, 2016-11-02:
Quick (and cold) flight test shows that all MAG transitions are now working great :)