iNavFlight/inav #718: Support multiple I2C buses at the same time

DzikuVx, 2016-10-23:
While working on #717 and #688 I've discovered that we do not initialize multiple I2C devices even if target requires that.
For example REVO target, built-in MAG and BARO usues I2CDEV_1 while I2C pins on board are connected to I2CDEV_2

digitalentity, 2016-11-15:
Initialisation of `I2C_DEVICE_EXT` came along with 0d4a3988929c7a0e6ba17a4bae28043e01a7b4a2

However sensor can be compiled only to use one or the other. I.e. on REVO we can have support either for onboard HMC5883L or the external one. Can't support both and switch by a CLI option.

DzikuVx, 2016-11-15:
OK, I'm closing this one since it solves the problem I've raised. Sensor assignment is completely different thing

closed 2016-11-15 by DzikuVx