iNavFlight/inav #729: Increase number of supported servos from 8 to 16

DzikuVx, 2016-10-28:
When PCA9685 is connected and configured, increase number of supported servos to 16

driver33, 2016-10-28:
I made changes in the code which works for me - however I am using only servos (smix) and not using motors (mmix)

Below are changes :
src/main/drivers/pwm_mapping.h :

#if defined(USE_QUAD_MIXER_ONLY)
#define MAX_PWM_MOTORS  4
#define MAX_PWM_SERVOS  1
#define MAX_MOTORS  4
#define MAX_SERVOS  1

#elif defined(TARGET_MOTOR_COUNT)
#define MAX_PWM_MOTORS TARGET_MOTOR_COUNT
#define MAX_PWM_SERVOS 16
#define MAX_MOTORS  TARGET_MOTOR_COUNT
#define MAX_SERVOS  16

#else
#define MAX_PWM_MOTORS  12
#define MAX_PWM_SERVOS  16
#define MAX_MOTORS  12
#define MAX_SERVOS  16
#endif

src/main/flight/mixer.h :
#if defined(USE_QUAD_MIXER_ONLY)
#define MAX_SUPPORTED_MOTORS 4
#define MAX_SUPPORTED_SERVOS 1

#elif defined(TARGET_MOTOR_COUNT)
#define MAX_SUPPORTED_MOTORS TARGET_MOTOR_COUNT
#define MAX_SUPPORTED_SERVOS 16

#else
#define MAX_SUPPORTED_MOTORS 12
#define MAX_SUPPORTED_SERVOS 16
#endif

src/main/flight/servos.h :
// target servo channels
typedef enum {
    SERVO_GIMBAL_PITCH = 0,
    SERVO_GIMBAL_ROLL = 1,
    SERVO_ELEVATOR = 2,
    SERVO_FLAPPERON_1 = 3,
    SERVO_FLAPPERON_2 = 4,
    SERVO_RUDDER = 5,
    SERVO_THROTTLE = 6, // for internal combustion (IC) planes
    SERVO_FLAPS = 7,
    SERVO_FLAPS2 = 8,
    SERVO_AUX1 = 9,
    SERVO_AUX2 = 10,
    SERVO_AUX3 = 11,
    SERVO_AUX4 = 12,
    SERVO_AUX5 = 13,
    SERVO_AUX6 = 14,
    SERVO_AUX7 = 15,
    SERVO_AUX8 = 16,

    SERVO_BICOPTER_LEFT = 4,
    SERVO_BICOPTER_RIGHT = 5,

    SERVO_DUALCOPTER_LEFT = 4,
    SERVO_DUALCOPTER_RIGHT = 5,

    SERVO_SINGLECOPTER_1 = 3,
    SERVO_SINGLECOPTER_2 = 4,
    SERVO_SINGLECOPTER_3 = 5,
    SERVO_SINGLECOPTER_4 = 6,

} servoIndex_e; // FIXME rename to servoChannel_e

DzikuVx, 2018-05-14:
I would like this  one to stay :)

DzikuVx, 2019-10-09:
This is already done for 2.3

closed 2019-10-09 by DzikuVx