iNavFlight/inav #735: NMEA support got GPS now and going forward.

nickgb, 2016-11-01:
I have been having problems getting iNav to recognise a GPS emitting NMEA messages.  The messages look good coming out of the GPS but I don't know how to tell if the iNav code (running on a NAZE32) is making any sense of them.  Is there a recommended way of debugging this kind of problem?  Is there anything critical needed by iNav such as a specific sequence for the messages?

Background:  I want to run several quadcopters in precision flight.  My test setup works well with a straight uBlox M8N emitting uBlox messages. I am trying to get the EMLID REACH RTK GPS to be recognised.  This has several options for output but the uBlox message protocol is not one of them.  NMEA is the common denominator.

I have seen some discussion about dropping NMEA from the cleanflight /iNav code to make space for more interesting functions.  Is this likely to be happening? soon?

DzikuVx, 2016-11-01:
NMEA support has been removed from Naze32 target due lack of flash space. Naze boards supports only UBLOX protocol. See https://github.com/iNavFlight/inav/wiki/Hardware-and-feature-support-map

closed 2016-11-01 by DzikuVx

stronnag, 2016-11-01:
NMEA is already removed for F1 targets due to space limitations.