iNavFlight/inav #743: Fixed-wing documentation

flounderscore, 2016-11-02:
I've been following the iNav development for I while now and just set it up on a flying wing successfully. I noticed that there is only little fixed-wing specific documentation and I would like to help (although I only have very little experience so far). iNav would probably a lot more popular in the FW community if it was better known that fixed-wing aircraft is supported, too.

The fixed-wing guide in the wiki is a good starting point, but seems to me a little outdated.

Also, there is general documentation both in the *.md files and in the wiki - somewhat redundant. What is the preferred place for up-to-date documentation?

Would it make sense for me to start a PR with a "template" for fixed-wing documentation with information currently on the wiki and mark sections were more experienced users need to chip in?

oleost, 2016-11-02:
All improvements are welcome.
Unsure what you want to write up on.

If its a guide on how to get iNav running on fixed-wing I believe the pre existing Wiki page is the right place to start, https://github.com/iNavFlight/inav/wiki/3.-Fixed-wing-guide

Along with addiotinal infomation regarding fixed wing in other pages where its seems fit.

Also unsure what parts of the Wiki is outdated with regards of Fixed wing?

flounderscore, 2016-11-03:
Things I've been missing: mention TPA for airplane, explanation what certain flight modes do for fixed wing, how to trim, and the new launch feature.

digitalentity, 2016-11-03:
@flounderscore any improvement is very welcome. You can have a look at https://github.com/iNavFlight/inav/wiki/PID-Attenuation-and-scaling, it's actually related to not yet merged https://github.com/iNavFlight/inav/pull/685, but you get the idea. Feel free to improve things!

oleost, 2016-11-03:
As he mentioned those are not in release version, not even in development branch.

The Wiki should reflect release version.

oleost, 2016-11-03:
But feel free to improve it

closed 2018-05-27 by stale[bot]