iNavFlight/inav #746: Extend MSP protocol for INAV settings

DzikuVx, 2016-11-03:
Apart from #709 there are several settings that can not be set via MSP. I've run through them and here they are:

| length    | setting                   | Payload Length    |
| ----      | ----                      | ----              |
| 1         | fixed_wing_auto_arm       | 1                 |
| 1         | nav_disarm_on_landing     | 2                 |
| 1         | nav_user_control_mode     | 3                 |
| 2         | nav_max_speed             | 5                 |
| 2         | nav_max_climb_rate        | 7                 |
| 2         | nav_manual_speed          | 9                 |
| 2         | nav_manual_climb_rate     | 11                |
| 1         | rth_alt_control_style     | 12                |
| 2         | nav_rth_altitude          | 14                |
| 2         | nav_fw_cruise_thr         | 16                |
| 2         | nav_fw_min_thr            | 18                |
| 2         | nav_fw_max_thr            | 20                |
| 1         | nav_fw_bank_angle         | 21                |
| 1         | nav_fw_climb_angle        | 22                |
| 1         | nav_fw_dive_angle         | 23                |
| 2         | nav_fw_loiter_radius      | 25                |
| 2         | flaperon_throw_offset     | 27                |
| 1         | flaperon_throw_inverted   | 28                |
| 2         | servo_lowpass_freq        | 30                |
| 1         | servo_lowpass_enable      | 31                |
| 2         | fw_iterm_throw_limit      | 33                |
| 1         | mode_range_logic_operator | 34                |
| 2         | acc_task_frequency        | 36                |
| 2         | attitude_task_frequency   | 38                |
| 1         | async_mode                | 39                |
| 2         | nav_landing_speed         | 41                |
| 1         | telemetry_inversion       | 42                |
| 1         | smartport_uart_unidir     | 43                |

Now, should we shorten the list, extenend it, or split into 2 frames.
We have free 128, 129, 130 and 131 frames

Thoughts? Personally I think this issue is quite important. I would even like to do it for 1.4 and I'm willing to implement Configurator changes

digitalentity, 2016-11-03:
We'll probably split it to at least 6-7 frames:

1) Position estimator settings (inav_*)
2) General navigation settings (nav_*)
3) Multicopter-specific navigation settings (nav_mc_*)
4) Airplane-specific navigation settings (nav_fw_*)
5) Airplane general settings (flaperon_*, fw_iterm_*, 
6) Gyro/PID settings (async_mode etc)
7) all the rest

What do you think?

digitalentity, 2016-11-03:
I think we need to keep #473 in mind and parameter groups require that you have one MSP message per one group of settings for most optimal implementation.

stronnag, 2016-11-03:
Useful. The ground stations could then offer infield adjustments of nav parameters.

DzikuVx, 2016-11-03:
It might be problematic to find that long bank of free message ids.

<https://github.com/iNavFlight/inav/issues/746#issuecomment-258118506>, or mute
the thread

Pozdrawiam,
Paweł Spychalski

digitalentity, 2016-11-03:
@DzikuVx MSP IDs
128-179 and 222-238 appear to be free

martinbudden, 2016-11-03:
I certainly agree we should split this into several frames. We should also try and stay compatible with betaflight/Cleanflight for the non-navigation settings, if at all possible.

Note that the MSP protocol is independent of parameter groups. That is, the MSP protocol will not be changed by converting to parameter groups.

However once we have converted to parameter groups, we can implement a new protocol using parameter groups - this will be more robust and easily extendable since parameter groups contain a version per group.

digitalentity, 2016-11-03:
@DzikuVx also IDs 6-9 and 12-33 are free to use.

@martinbudden original idea of PGs were to have same data structure in EEPROP, in RAM and on the wire via MSP. So while aligning MSP and PGs is not required, it's preferred.

DzikuVx, 2016-11-03:
OK, I will restructure and give proposal here in some time

digitalentity, 2016-11-03:
@DzikuVx I'm already restructuring NAV-specific settings. Will submit a PR in a few, you can take it from there.

DzikuVx, 2016-11-03:
@digitalentity OK

digitalentity, 2016-11-03:
@DzikuVx see #747

DzikuVx, 2016-11-03:
This is my proposal for first of new frames. It hold INAV specific async mode and PID controller settings not covered in other frames, including #709.

Payload is 15 bytes long including reserved 6 bytes for future usages.

**MSP_INAV_PID**

Frames:

* MSP_INAV_PID, Frame ID _6_
* MSP_SET_INAV_PID, Frame ID _7_

| length    | setting                   | Payload Length    |
| ----      | ----                      | ----              |
| 1         | async_mode                | 1                 |
| 2         | acc_task_frequency        | 3                 |
| 2         | attitude_task_frequency   | 5                 |
| 1         | mag_hold_rate_limit       | 6                 |
| 1         | MAG_HOLD_ERROR_LPF_FREQ (not implemented yet as configurable) | 7     |
| 2         | yaw_jump_prevention_limit | 9                 |
| 6         | _reserved_                | 15                |

@stronnag @martinbudden @digitalentity thoughts?

digitalentity, 2016-11-03:
@DzikuVx thumbs up!

DzikuVx, 2016-11-03:
MSP_INAV_PID docs updated, payload kept at 15 bytes: I've forgot about gyro_lpf :)

MSP_INAV_PID will be done as #709 there is just too much to divide it into steps

DzikuVx, 2016-11-03:
MSP_INAV_PID and MSP_SET_INAV_PID implemented in #709

DzikuVx, 2017-01-05:
MSP_NAV_POSHOLD in #1002

DzikuVx, 2017-01-24:
Since this topic is now active and I'm adding new MSP frames from release to release, I'm closing this ticket. New frames will be added separately from now

closed 2017-01-24 by DzikuVx