iNavFlight/inav #750: iNav 1.2.1 Weird Elevon movements when throttle applied

moggiex, 2016-11-03:
Hi,

I can pop this in the RC groups thread if desired but seemed best to post here.

See this video https://www.youtube.com/watch?v=fEl_NIaN8_o

Watch the elevons, thats NOT right?!!! No wonder she stuffed in today when I applied angle mode.

Happens in both Angle and Horizon modes.

Settings dump here http://pastebin.com/Libw7GDv als I do have black box data, but not willing to post in a public forum (GPS data).

Is this my screw up or something else?

Matt

oleost, 2016-11-03:
For me it's looks normal, when you throttle up the integral part of the controller gets active and working.

Looking at the dump it looks you have not aligned your FC for leveled fight.

set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0

oleost, 2016-11-03:
So when your plane is flat, the flight controller think the plane is pointing upwards and tries to correct it

moggiex, 2016-11-03:
Howdy Oleost, firstly thank you for getting back to me so quickly, looking into this right now. I appreciate the help. Matt

moggiex, 2016-11-03:
Hi Oleost, checking the wiki here for "Level calibration" https://github.com/iNavFlight/inav/wiki/4.-Sensor-calibration it wasn't needed as I put that board in as closely to the center axis as I could. So they should be all 0? Matt

oleost, 2016-11-03:
So when your plane is straight in the air iNav configurator shows zero on all axis? 

Also when you move the airplane the airplane object moves the correct way?

moggiex, 2016-11-03:
Oleost, just read your reply again, watch closely, it appears the PID loop is feeding/bleeding throttle back into the elevon movements to the point they reach maximum deflection. That's definitely not right :) Matt

moggiex, 2016-11-03:
> So when your plane is straight in the air iNav configurator shows zero on all axis? Also when you move the airplane the airplane object moves the correct way?

It does, but it's irrelevant, watch when I apply throttle, the elevons move to the point of maximum deflection. Take the throttle off, they go back. 

Matt

oleost, 2016-11-03:
No it's not irrelevant. Point your airplane to the ground and I expect your elevons to go the opposite way.

When you apply throttle the flight controller assumes you flying. If the airplane is pointing down just small angle it will try and correct for it when your in angle mode, when nothing happens it will just try and correct more and more.

Also you should update to 1.3 as this limits the maximum throw I gain is allowed to correct for, but it will still move the elevons .

oleost, 2016-11-03:
When you have no throttle only P function is active. I functions activates when you apply throttle

moggiex, 2016-11-03:
Hi oleost, I appreciate that throttle should be accounted for in the i loop, but that is a swamping of the movements. IE when more throttle applied, smaller movements are required (should in theory be a greater air speed so less movement required), again check the video, even bringing it back straight the elevons don't correct themselves, they're swamped. 

I appreciate all the hard work on iNav oleost, but to flash the board to 1.3 (which was only released yesterday and sounds like it has a similar loop) I have to remove it out of the model (huge faff), if I'm doing that I'm going to switch to another Flight controller instead. Harsh, but that's my honest feelings right now. 

Sorry. I did come back & try it again in a new model with an update board as promised! Matt

oleost, 2016-11-03:
Both me, others and the main developer if iNav have success with iNav on wing.

"even bringing it back straight the elevons don't correct themselves, they're swamped."

As I said, this has been fixed in iNav 1.3. I gain is now being limited. That said it should not be an issue in flight, more at takeoff and bench testing in stabilise modes.

"but to flash the board to 1.3 (which was only released yesterday and sounds like it has a similar loop) I have to remove it out of the model (huge faff)"

To build your airplane without USB access to the flight controller is one major flaw.

"IE when more throttle applied, smaller movements are required (should in theory be a greater air speed so less movement required), "

This is true, we have TPA for that, which is improved regards to fixed wing in upcoming 1.4. Also pitot sensor is being worked one which in turns is much more accurate than relying on throttle. ( TPA is default turned OFF )

BUT this has nothing to do with what you are experiencing. I gain is completely shutoff with zero throttle to prevent the I gain to wind up while being stationary.

digitalentity, 2016-11-03:
> even bringing it back straight the elevons don't correct themselves, they're swamped.

Actually elevons shouldn't get back when you level the plane by hand. From FC point of view there is no "hand" and no "bench", it's flying and any airplane motion is natural. 

When you level the plane on the bench, the only thing FC knows is that plane is now "flying" with seriously deflected elevons and it's level. Again - as far as FC is concerned the plane **is "flying" level**, so it doesn't see a reason to change anything in elevon position.

This is how stabilisation works. The flight controller tried as hard as possible to reach the desired orientation and once it's reached - airplane stays there.

> but to flash the board to 1.3 (which was only released yesterday and sounds like it has a similar loop) I have to remove it out of the model (huge faff)

I agree with @oleost, having FC hidden inside a model without access to USB is not a very good idea. Sometimes critical bugfixes are released and you won't be able to use them without pulling your build apart.

> IE when more throttle applied, smaller movements are required (should in theory be a greater air speed so less movement required)

In 1.3 you have TPA which does the job and in 1.4 you'll have even better TPA. Also, thanks to @sambas we'll soon have pitot sensor support and even more robust PID scaling according to real airspeed.

digitalentity, 2016-11-03:
The flying wing behavior is tested for upcoming 1.4 by myself. @oleost and probably some other people with success. I believe this issue is irrelevant anynore.

closed 2016-11-03 by digitalentity