iNavFlight/inav #753: Support Sonar on SP Racing F3 EVO

kitchung, 2016-11-04:
CleanFlight 1.14.0, released yesterday enabled Sonar support for the SP Racing F3 EVO. Please consider adding support in iNav. Please see https://github.com/cleanflight/cleanflight/commit/6b03f97c522442b27b3c480ab79a2d85ded85366

DzikuVx, 2016-11-04:
@kitchung do you own EVO and are you willing to test it?

kitchung, 2016-11-05:
Yes, I am building a new quad with an EVO board now with all necessary parts (board, HC-SR04 sonar module, 1k resistors). Initial testing will be bench testing.

martinbudden, 2016-11-05:
Closing. Addressed by PR https://github.com/iNavFlight/inav/pull/759

closed 2016-11-05 by martinbudden

kitchung, 2016-11-06:
FYI bench test of sonar on SP F3 EVO with Clean flight 1.14 was successful. I'll test iNav as soon as binaries are available.

martinbudden, 2016-11-06:
Great, thanks for testing.

kitchung, 2016-11-06:
Well, that was for CleanFlight 1.14; will test iNav as soon as binaries are available.

oleost, 2016-11-07:
@kitchung You could setup your own build eviroment, tbh its really quick and it was you that requested this.

Anyway, here is latest developement hex for evo. use with caution. https://drive.google.com/open?id=0B-yHzNtOXxbEM0hYcnhUUzBPOFU

kitchung, 2016-11-11:
@oleost  Thanks for the binaries. Bench testing was successful!

neKraehe, 2017-03-09:
@martinbudden @kitchung  Hello! I would like to test out Sonar on the SP F3 EVO aswell, what are the pins that I need to use and will I need the 1k resistors with them? What are the needed settings in the gui for that? Thanks in forward!

DzikuVx, 2017-03-09:
@neKraehe Sonar support is disabled in INAV 1.6 on all targets