iNavFlight/inav #767: VCP not stable on F4 boards

digitalentity, 2016-11-06:
Encountered on Pixracer target (#466) rebased on development (a3b81e36e2dcc78ddc74877852d093c4c795e34a).

VCP connection to Configurator freezes after a while. Board CPU is very busy in `VCP_Ctrl` and `usbVcpFlush`. F1 (CC3D) and F3 (EVO) targets don't seem to be affected.

Can anybody confirm the issue on their F4 boards?

digitalentity, 2016-11-06:
Interestingly, issue is resolved with disabling LTO. Another instance of (in)famous memory corruption bug?

DzikuVx, 2016-11-06:
tested with Flip32 F4, no problems

jflyper, 2016-11-07:
> Interestingly, issue is resolved with disabling LTO. Another instance of (in)famous memory corruption bug?

In betaflight, I've experienced the opposite case; disabling LTO wedges F4 targets upon entering CLI: betaflight/betaflight#1454.

digitalentity, 2016-11-10:
I'll keep it open, but low priority

closed 2016-12-14 by digitalentity