iNavFlight/inav #768: Deprecate gyro_sync and looptime

digitalentity, 2016-11-06:
90% of the boards out there have gyro data ready interrupt and are capable of `gyro_sync = ON`.
How about we deprecate gyro_sync (make it always on)? This will also make looptime obsolete.
With #421 merged we can define all rates as denominators:

`gyro_denominator` - denominator for gyro read, relative to hardware sampling rate
`acc_denominator` - acc update denom, relative to gyro
`atti_denominator` - attitude denom, relative to gyro
`pid_denominator` - PID process denom, relative to gyro

Basically, gyro is the fastest,  the rest are 2,3,4... times slower.

For `async_mode = NONE` acc_denominator, atti_denominator and pid_denominator are ignored.

digitalentity, 2016-11-06:
@DzikuVx, @martinbudden what do you think?

DzikuVx, 2016-11-06:
I'm fine with gyro_to_pid_denominator. It makes sense. But it really makes sense only with gyro_lpf=256HZ. 1kHz step does not gives much flexibility. 

But I do not see any benefit of coupling accelerometer and attitude with gyro again.

martinbudden, 2016-11-06:
Yes, I'm in favour of deprecating `gyro_sync`. Betaflight did it ages ago.

DzikuVx, 2016-11-06:
In terms of assuming we always sync to gyro, I'm also in favour of it

digitalentity, 2018-05-14:
Irrelevant for a while now

closed 2018-05-14 by digitalentity