iNavFlight/inav #774: Detecting MS5611 baro falis on SPRacingF3 (Acro) board

Xe54, 2016-11-08:
Powering the board causes an I2C error with the current version 1.3 and the baro is not activated in the hardware bar. After rebooting the board (switching to cli and back) the baro is detected and works properly. Known workaround is to explicitly set baro_hardware = 3. Here is the status before workaround:

# status
System Uptime: 94 seconds, Voltage: 156 * 0.1V (4S battery - OK), System load: 0.10
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 1009, I2C Errors: 2, config size: 1988

And here is the status after:

# status
System Uptime: 22 seconds, Voltage: 156 * 0.1V (4S battery - OK), System load: 0.10
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 1009, I2C Errors: 1, config size: 1988

Xe54, 2016-11-08:
My dump:
# dump

# version
# INAV/SPRACINGF3 1.3.0 Nov  2 2016 / 11:15:52 (479d607)
# pflags
# Persistent config flags: 0x00000001
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -UNUSED_1
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -UNUSED_2
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature SERVO_TILT
feature SOFTSERIAL
feature GPS
feature FAILSAFE
feature TELEMETRY
feature CURRENT_METER
feature BLACKBOX

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
[... 373 lines cut ...]
set max_angle_inclination_pit = 500
set gyro_soft_lpf_hz = 60
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 80
set rc_yaw_expo = 40
set thr_mid = 45
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

digitalentity, 2016-11-21:
@Xe54 can you verify with INAV 1.4-RC1?

DzikuVx, 2016-11-21:
I can test all my baros today with SPRF3

Xe54, 2016-11-23:
I will give it a try this weekend.

DzikuVx, 2016-11-24:
Just testes SPRF3 Acro on INAV 1.4RC1 with all my baros: MS5611, BMP085/180, BMP280

In all cases baros were detected without any problems. Closing. @Xe54  if problem will reappear, reopen this ticket

closed 2016-11-24 by DzikuVx

Xe54, 2016-11-26:
Sorry guys I don't know if i'm doing something wrong, but it got worse. Now i'm getting four I2C errors more. My guess is, the problem is still there, but due to this bugfix: "Fixes for sensor detection, now INAV will retry sensor detection up to 5 times before reporting it missing." it is displayed five time now. Setting baro_hardware to 3 still help. Other than before is the behavior of the baro. It is displayed from the beginning in the hardware bar and seems to work (i couldn't perform a flight test yet).

The status below show the errors before and after setting baro_hardware to 3:

# status
System Uptime: 509 seconds, Voltage: 163 * 0.1V (4S battery - OK), System load: 0.09
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 1012, I2C Errors: 6, config size: 2024

# status
System Uptime: 212 seconds, Voltage: 163 * 0.1V (4S battery - OK), System load: 0.09
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 1005, I2C Errors: 1, config size: 2024

stronnag, 2016-11-26:
The difference of 5 is due to 5 failed probes for BMP085/BMP180 with baro_hardware=0 and is entirely to be expected. Explicitly setting baro and mag hardware is encouraged for 1.4.

So it has not got worst, it is working as designed.

From both examples it appears that the baro is detected.

Xe54, 2016-11-26:
@stronnag I doubt it is the desired behavior of the hardware auto detecting loop. It is not very user friendly if each piece of hardware produces tons of errors at initialization and provides the user to seek the source of those errors. But if it so, then I would suggest to completely disable auto detecting and prompt the user to enable hardware manually.

stronnag, 2016-11-26:
Actually it is the intention, however sub-optimal this may be. The affect of user confusion was predicted at the time of commit. Thank you for confirming that prediction. 

See the final comments in https://github.com/iNavFlight/inav/pull/801#issuecomment-261764676 and https://github.com/iNavFlight/inav/pull/809#issuecomment-261895253