iNavFlight/inav #775: Failsafe not working on development.

oleost, 2016-11-09:
Hi. First I want to say I have not gotten to test this on iNav 1.3 yet because of time limitation.

Im having some issues getting Failsafe RTH working reliable on my fixed wing.

Made and video while I did several attempt.

Only LOG25 and around 5 minutes in the video it works. I have not found the reason for it working that time and not the other.

Dump, and logs and video of all attempts are in this link: https://drive.google.com/open?id=0B-yHzNtOXxbEX1dxSTZod3lnT0U

I will try to do so more backtracking and figuring out whats going on, but if someone have some ideas what to test, or what can be the cause it would be great. @digitalentity ?

oleost, 2016-11-09:
[image]

[image]

oleost, 2016-11-09:
Noticed on the LOG25 it says BARO, while on this board I have baro disabled.

Also noticing on my telemetry gAlt tells the actually altitude. Like 30m above sea level, and not from the point of arming.

Is there a bug regarding using Gps altitude only compared to baro?

digitalentity, 2016-11-09:
Blackbox explorer doesn't show correct values at the moment.
`failsafePhase = 2` is actually `RETURN_TO_HOME`, not "LANDING"
`BARO` is actually `NAV ALTHOLD`

digitalentity, 2016-11-09:
@oleost how are you testing failsafe? Turning off your TX?

oleost, 2016-11-09:
@digitalentity Yes, turning off TX

There was only 1 time where I screwed up and disarmed instead 👎  But then I said so in video

digitalentity, 2016-11-09:
In most of the logs navState is stuck at 8 (RTH_INITIALIZE)

This is a synergistic effect of two factors:
1) You are not flying the airplane (speed is low) and you don't have a compass = true heading is unknown so no RTH can be executed as INAV doesn't know where airplane is going
2) Actual bug that doesn't allow the RTH_INITIALIZE state to be re-checked. The bug was introduced by #745.

Fix pending.

PR #776 (2016-11-10): Fix RTH_INITIALIZE state getting stuck in some cases
Forgot to set up the timeout event in #745. Fixes #775.
closed 2016-11-16 by digitalentity