iNavFlight/inav #778: Disable disarming when failsafe is activated as a flight mode

digitalentity, 2016-11-09:
When RC receiver only supports setting channel values on failsafe the only way to engage failsafe procedure is to configure `FAILSAFE` box mode.

The problem is that if one uses switch arming, the machine will disarm on failsafe if proper channel is not set to `ARM`. This is, however, unsafe, because it will make the machine actually arm when you turn off the TX.

Proposed fix - retain arming state when failsafe is enabled via `FAILSAFE` box mode. This way on failsafe maching won't arm or disarm and failsafe procedure will be correctly executed.

digitalentity, 2016-11-09:
Interestingly, code already has protection against disarming by a switch when failsafe is active. I need to verify that it's working correctly.

oleost, 2016-11-09:
Remember it also need to work with throttle channel set to failsafe 1000us / 0%

digitalentity, 2016-11-10:
Tested with all throttle positions and both no-pulses failsafe and `FAILSAFE` mode with set-channels failsafe. All working as expected.

PR to fix the issue is in place. Closing.

PR #789 (2016-11-15): Fix failsafe autodisarm issues
Fixes #778, #786
closed 2016-11-10 by digitalentity