iNavFlight/inav #784: bug in Soft I2C driver

night-ghost, 2016-11-10:
There is a bug in
```
static bool I2C_Start(void)
{
    SDA_H;
    SCL_H;
    I2C_delay();
    if (!SDA_read) {
        return false;
    }
    SDA_L;
    I2C_delay();
    if (SDA_read) {
        return false;
    }
    SDA_L;
    I2C_delay();
    return true;
}

```

should be 
```

    SCL_L;
    I2C_delay();
    return true;
}
```

DzikuVx, 2016-11-10:
@digitalentity ??

digitalentity, 2016-11-10:
Yes, this is a bug. Will fix, however we don't use soft-i2c anywhere

digitalentity, 2016-11-10:
PR in place. Closing

closed 2016-11-10 by digitalentity

night-ghost, 2016-11-10:
>however we don't use soft-i2c anywhere

Soft_i2c is much more reliable than SoftSerial so on some boards it is better to move external i2c to Input port via it, and to use the FlexiPort as a 2nd hardware serial.

digitalentity, 2016-11-10:
> Soft_i2c is much more reliable than SoftSerial so on some boards it is better to move external i2c to Input port via it, and to use the FlexiPort as a 2nd hardware serial.

Hm... didn't think about it from that perspective. Also soft-i2c consumes much less memory compared to softserial.

digitalentity, 2016-11-10:
See #788 to track the idea.