iNavFlight/inav #786: BUG: If failsafe is initated without throttle it will not work.

oleost, 2016-11-10:
This is on almost latest development.

The case:

If Stage 1 throttle is set to auto the failsafe wont work.
If Stage 1 throttle is set to hold, the failsafe will work IF you have throttle applied from the TX when loosing signal / going into failsafe.

https://drive.google.com/open?id=0B-yHzNtOXxbERkUwVHJuS3czVDA

3 logs and cli dump.

On all three logs TX sticks want to cmmand 1700us, but motor output is only 1000us except on LOG0038 when both throttle is set to hold and throttle is applied.

oleost, 2016-11-10:
Does #789 also fix this?

digitalentity, 2016-11-10:
Yes, but with a limitation - if your throttle is LOW for longer then `failsafe_throttle_low_delay` it will disarm.

digitalentity, 2016-11-10:
We need a possibility to disable this feature - "disarm instead of failsafe when throttle is LOW for too long".

digitalentity, 2016-11-10:
Will fix this as a part of #789 as well

digitalentity, 2016-11-10:
Fixed. Wiki and docs need updating. Now `failsafe_throttle_low_delay = 0` means "don't disarm when failsafe at low throttle".

oleost, 2016-11-10:
Great. Will do some testing in a couple of hours.

oleost, 2016-11-10:
Btw with all my test I had throttle cut shorter period than failsafe_throttle_low_delay = 100

oleost, 2016-11-16:
Fixed as a part of#789

closed 2016-11-16 by oleost