iNavFlight/inav #788: Use SOFT_I2C on some targets

digitalentity, 2016-11-10:
On targets where UART is shared with I2C it's more wise to disable `softserial`, enable `soft_i2c` on those pins and leave Flexi port as a normal UART.

Let's discuss this.

night-ghost, 2016-11-10:
sorry not "on those pins" - but on some else pins. Good candidates are unused pins on Input port.

I check this driver in my fork of Ardupilot for RevoMini, and the results are very pleased me - the number of bus resets with this (slightly reworked) driver is noticeably smaller than with the hardware one.

digitalentity, 2016-11-10:
@night-ghost INAV will benefit from your reworked driver. Can you share the code or maybe make a PR directly?

night-ghost, 2016-11-10:
It is reworked not by logic but to work in absolutely different driver model. So Inav don't requires this :)

Some days later I'll push recent changes and 'll add comment

digitalentity, 2016-11-10:
@DzikuVx @oleost @DzikuVx I'd like to hear your thoughts about this feature.
Should we create special "flavours" of CC3D and some others moving I2C away from Flex port, leaving Flex port Serial-only and removing SoftSerial?

DzikuVx, 2016-11-10:
Well.... I checked the numbers. CC3D users are more less 6% of our users and number is going down. We agreed not to drop support for F1 boards yet, but support does not means we should invest quite a lot time only to develop features that very limited number of users will ever use. And the number will go down every month since well, you know... 
If I remember correctly CC3D already has 2 flavours.
But, if someone, for example @night-ghost  offers to bring this solution I have nothing against it.

martinbudden, 2016-11-10:
A year ago I would have said yes. I still really like the CC3D, but I think the main limitations now is the ROM size rather than the number of output ports. I think we should allow users to build additional flavours if they want, but not actually release them ourselves.

night-ghost, 2016-11-10:
>But, if someone, for example @night-ghost offers to bring this solution I have nothing against it.

Your assumption is wrong -  I'm talking about the Revolution board with processor STM32F405

DzikuVx, 2016-11-10:
@night-ghost Ah... that changes everything. Possibility to have both USART3 and I2C would be very nice.

oleost, 2016-11-10:
Isn't there possible to do this without special flavour's?

Like a setting

Output_mode = softi2c or softserial?
And then having main UART always as UART?

DzikuVx, 2016-11-10:
@oleost on F4 there should be no problem with such approach.

oleost, 2016-11-10:
F4 I'm talking about. I also agreeing with not spending time doing a lot of work in aging F1 targets which will only benefit very few people.

F3 is very cheap now, so most new builds will have those.
Current builds have cc3d etc and they only needs new fixes and features, of iNav itself. At least my opinion.

closed 2017-10-29 by digitalentity