iNavFlight/inav #791: Parachute in configurator doesn't show SBUS failsafe

mmormota, 2016-11-12:
SBUS has a bit to indicate failsafe condition (no need to special failsafe channel values etc).
iNav seems to detect this flag,  for example:
- it sets throttle to minimum in the receiver menu, and other stick channels to middle
- follows the "channel fallback settings" in the failsafe menu
But the small parachute on the top not indicates the detected failsafe condition.

oleost, 2016-11-12:
Not at home now so I can't verify.

But I'm 99% sure my parachute lights up when I turn off my transmitter.

I got and Frsky XSR with sbus set up with "No Pulses"

mmormota, 2016-11-12:
SBUS is a serial protocol, I suppose "no pulses" is about ppm or cppm. The serial data stream is flowing during failsafe (I checked by a scope). There are bits in the sbus packet for "signal ok", "signal lost" and "failsafe" condition. I didn't looked into the code, but suppose that iNav checks these bits in case of sbus input.

mmormota, 2016-11-12:
According to my efforts, the parachute lights up, if failsafe is assigned to a channel, and the channel value enters into the defined range.

closed 2017-01-30 by digitalentity