iNavFlight/inav #794: CC3D - new in-out order

hali9, 2016-11-13:
Is it possible to change the order of inputs and outputs, to connect more servos?
For example:
S6_OUT is not use and have adc - It could be used by releasing the single in-pin.
Next timer2 could be used to softserial for pins(S5_IN, S6_IN).
If the PPM will be on IN_1 we have all 4 channels timer3 free and it can be use for 4 servos or other feature like RSSIADC, Current, LEDStrip (if passible) and beeper (if passible, if not passible then PA15).

S1_IN (PB6) - TIM4_CH1 -> PPM 
S2_IN (PB5) - TIM3_CH2 -> TX/SonarTriger, change to S2/Beeper?
S3_IN (PB0) - TIM3_CH3/ADC12_IN8 -> RX/SonarEcho, change to S3/RSSIADC
S4_IN (PB1) - TIM3_CH4 ADC12_IN9 -> Current change to S4/Current
S5_IN (PA0) - TIM2_CH1 ADC12_IN0 -> Vbattery change to  RX/SonarTriger
S6_IN (PA1) - TIM2_CH2 ADC12_IN1 -> RSSIADC change to TX/SonarEcho

S1_OUT (PB9) - TIM4_CH4 -> M1
S2_OUT (PB8) - TIM4_CH3 -> M2
S3_OUT (PB7) - TIM4_CH2 -> M3 
S4_OUT (PA8) - TIM1_CH1 -> M4 (S1 - serwo tilt), change to M4
S5_OUT (PB4) - TIM3_CH1 -> LED/M4 (serwo tilt), change to S1\LED
S6_OUT (PA2) - TIM2_CH3 ADC12_IN2 -> not used, change to Vbattery

My idea is that if the corresponding feature (RSSI, current, LED strip) in inav-configurator is switched off, and if servo_tilt feature is on then servo is on this pin and can be configure on servo tab in inav-configurator.

Sorry for my poor English, I tried to write clearly.

digitalentity, 2016-11-21:
Don't think quad only, INAV also supports Tri, Hexa, Airplane (which uses all 6 output). With your approach we'll break some of the critical functionality.

hali9, 2016-11-21:
Ok. but it is cc3d which does not support hex'a, max motor is 4 (TARGET_MOTOR_COUNT 4 in CC3D\target.h). And airplane have its own config airPPM diffrent from multiPPM (CC3D\target.c).

But if you see my proposal, you can have up to 8 output (4 motors and 4 servos). 
And servos use its own 4 channel timer3 and can be replace all on 4 motors.
It means you could have a hex or octocopter on CC3D. 

Unless I'm wrong?

digitalentity, 2016-12-14:
PPM input is on S6_IN for compatibility with LibrePilot. We are not going to change it back to S1_IN. I also don't see much sense in hacking CC3D anymore. F1 boards are EOL, we are supporting them as long as they are used, but they are not recommended for newer setups.

closed 2017-01-05 by DzikuVx