iNavFlight/inav #798: Speed in CRUISE mode limited by the smaller of nav_max_speed, nav_manual_speed

miradarya, 2016-11-15:
Reproduced easily in 1.3, e.g. with default settings nav_max_speed=300 and nav_manual_speed=500. Vehicle will go 300cm/s instead of 500cm/s in CRUISE mode and full stick deflection. My coding skills have been resting for quite a while, but I think I found the reason in the code. While target position gets moved by nav_manual_speed in adjustMulticopterPositionFromRCInput, speed is generally limited by nav_max_speed in updatePositionVelocityController_MC (as a result of getActiveWaypointSpeed). Not sure if this is the intended behavior.

digitalentity, 2016-11-21:
This is the intended behavior.
`nav_max_speed` defines the global speed limitation that can't be breached by any flight mode.
`nav_manual_speed` imposes an additional limitation for manual control.

closed 2016-11-21 by digitalentity

giacomo892, 2017-04-08:
Is they any way to open this again?

From documentation:

nav_max_speed = Maximum velocity firmware is allowed in full auto modes (POSHOLD, RTH, WP) [cm/s] [Multirotor only]

nav_manual_speed	=Maximum velocity firmware is allowed when processing pilot input for POSHOLD/CRUISE control mode [cm/s] [Multirotor only]

It seems like nav_manual_speed can be more than nav_max_speed  as written in documentation and I think is nice to have a maximum limit for fully automatic flight and another limit for manual gps crusing (nav_manual_speed	) which can be higher than nav_max_speed .

Can we implement this behaviour?

digitalentity, 2017-04-08:
This behavior already changted by https://github.com/iNavFlight/inav/pull/1401

giacomo892, 2017-04-08:
I've setted my manual speed to 1400 and nav_max_speed to 600 and my quad was stuck to 18km/h in GPS CRUISE (it was slow, since it was intended to do 50km/h).
 
I try again in some minutes and report with a log if things aren't working as expected.

Thanks